maximkulkin/esp-homekit #1: Add configuration option for custom / generated password

maximkulkin, 2017-12-06:
Right now password is hardcoded to be "111-11-111". This should be configurable: developer should be able to either provide a custom password or specify that random password should be generated each time. In later case, there should be a configuration option to provide password callback, so that random password could be exposed to user during pairing process.

HomeACcessoryKid, 2017-12-09:
make overload parameters in the home kit_server_init so you can set things like password, category and more. At evaluation moment, if value is not set, make it the default.
My AiLight contains an example of setting the accessory category like this

HomeACcessoryKid, 2017-12-09:
another option is that the homekit_accessory_t becomes a member of homekit_config_t which can also carry other config items as above

maximkulkin, 2017-12-11:
Done

closed 2017-12-11 by maximkulkin