maximkulkin/esp-homekit #3: mDNS reconfiguration

maximkulkin, 2017-12-06:
When accessory starts blank slate, it starts in pair setup mode allowing any controller to pair with it. mDNS TXT has "sf=1" record. After initial pairing, pair setup mode should be disabled, accessory should refuse to pair and mDNS TXT should have "sf=0". Vise versa, after removing last (admin) controller, accessory should switch back to pairing mode and announce that through mDNS.

Unfortunately, current mDNS library does not support run-time reconfiguration and should be extended/replaced to support that. It seems that "lwip" library already has support for mDNS and it would be beneficial to reuse that one instead of custom library that we use now.

Task is to
1. implement mDNS responder based on lwip library
2. add calls to mDNS reconfiguration when pairing is done/removed

closed 2017-12-24 by maximkulkin

Rjayone, 2019-07-20:
Hi @maximkulkin, there is any updates about mdns reconfig? I was faced with problem, when removed the accessory. Now the device announced with status sf=0

maximkulkin, 2019-07-22:
@Rjayone It usually works fine. Are you sure you did it the right way (e.g. actually being on the same network when removing accessory) ? If not, accessory won't know you have removed it.

Rjayone, 2019-07-23:
I removed the house from Home at all. I'm not sure it was a correct way. Also, i'm not sure that i used latest lib
So, there is no any (easy) ways to reset the device when it was removed, not being in the same network?

maximkulkin, 2019-07-23:
Depends on which firmware you have used. Some firmwares support complete config reset when you long press a button. If you do not have that, an easy way is to erase and reupload firmware. Erasing flash will erase sector that holds HomeKit config. You could also do that by erasing just the region with config with following command `esptool.py -p /dev/tty.SLAB_USBtoUART -b 115200 erase_region 0x7a000 4096` (assuming /dev/tty.SLAB_USBtoUART is your serial port, 0x7a000 is your HOMEKIT_SPI_FLASH_BASE_ADDR... Change those according to your setup)