maximkulkin/esp-homekit #5: #define SPIFLASH_BASE_ADDR 0x200000

HomeACcessoryKid, 2017-12-10:
Hi,

The use of the area from 2Mbyte up is not compatible with the use of devices build by Sonoff which are probably our biggest target for real life deployment. Personally I would use 0x7A000 as being compatible with such devices and allowing FOTA strategies

maximkulkin, 2017-12-10:
I’m not an expert in memory layout, this address seem to work for me. But! As with any flash library it can be configured with preprocessor defines in Makefile.

nickw444, 2017-12-29:
Also had trouble with this, setting it via `EXTRA_CFLAGS` didn't seem to work for me, had to go and manually set it in `component.mkL15`: `HOMEKIT_SPI_FLASH_BASE_ADDR ?= 0x70000`.

(Using this with an ESP-01)

maximkulkin, 2018-01-08:
Ok, for the record so that people who will read it later won't get wrong information:
to use example on ESP-01, you need to set FLASH_SIZE variable to 8 and HOMEKIT_SPI_FLASH_BASE_ADDR to 0x7a000 (and of course latest version of the code):

```
make -C examples/led FLASH_SIZE=8 HOMEKIT_SPI_FLASH_BASE_ADDR=0x7a000 all
```
or through environment variables:
```
export FLASH_SIZE=8
export HOMEKIT_SPI_FLASH_BASE_ADDR=0x7a000
make -C examples/led all
```

closed 2018-01-17 by maximkulkin