maximkulkin/esp-homekit #10: Crashes when additional client connects to accessory

maccoylton, 2017-12-22:
HI, I have my accessories shared with a some of the family, seems when any of those additional clients connects, my accessory crashes:- 

target temperature set to : 17.000000
Target set to on
>>> HomeKit: [Client 4] Update Characteristics

target temperature set to : 16.500000
Target set to on
>>> HomeKit: [Client 4] Update Characteristics

target temperature set to : 16.000000
Target set to on
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics

target heating cooling state set to : 3
Target set to auto
Target Auto, current now off, current temp 19.500000, target temp 16.000000
>>> HomeKit: Got new client connection: 5
>>> HomeKit: [Client 5] Pair Verify Step 1/2
>>> HomeKit: [Client 5] Pair Verify Step 2/2
>>> HomeKit: [Client 5] Found pairing with 856FCE22-3DA1-4066-A021-BC1A1DC64344
>>> HomeKit: [Client 5] Verification successful, secure session established
>>> HomeKit: [Client 5] Get Accessories
Fatal exception (28): 
epc1=0x4000bf80
epc2=0x00000000
epc3=0x4020349d
excvaddr=0x00000000
depc=0x00000000
excsave1=0x40002ad0
Registers:
a0 40002ad0 a1 3fffc3e0 a2  00000000 a3  fffffffc
a4  000000ff a5  0000ff00 a6  00ff0000 a7  ff000000
a8  0000001f a9  00000022 a10 00000002 a11 0000005c
a12 000000c8 a13 00000000 SAR 00000018

Stack: SP=0x3fffc3e0

Free Heap: 4796
_heap_start 0x3fff1ba8 brk 0x3ffff388 supervisor sp 0x3ffffb00 sp-brk 1912 bytes
arena (total_size) 55264 fordblks (free_size) 2884 uordblocks (used_size) 52380

maximkulkin, 2017-12-22:
Yeah, I've seen that problem. Looks like cJSON library we use to build JSON responses is not very memory efficient (accessories JSON for Thermostat on my tests takes ~16K, then you need space for buffer to render it into string). ESP with my HomeKit library has ~27K. Also, each new client is handled in a new task, which also takes ~4K.
So, going forward I'm going to investigate other options for rendering JSON (maybe writing my own, more memory efficient library). Also, I will push a patch that I had for a while which modifies server to handle all clients in a single task which should improve memory requirements for each client. Each client still requires a separate 1K buffer for decoding data though.

maximkulkin, 2017-12-22:
BTW, if you'd like to join our (private) Slack channel, send me an email and I will send you invite

maximkulkin, 2017-12-27:
Should be fixed now: I rewrote JSON generation with a much more memory efficient one.

closed 2017-12-27 by maximkulkin