maximkulkin/esp-homekit #9: Issue Pairing - pair remove event

maccoylton, 2017-12-22:
Having successfully built and flashed a  couple of examples, I am now having some issue with pairing. Seems immediately after the verification  step is complete, and the get accessories is called, I then get a remove pairing event, summary and detailed traces below, any ideas?

>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
>>> HomeKit: [Client 4] Pair Setup Step 3/3
>>> HomeKit: Added pairing with 856FCE22-3DA1-4066-A021-BC1A1DC64344
>>> HomeKit: Configuring mDNS
>>> HomeKit: [Client 4] Successfully paired
>>> HomeKit: [Client 4] Closing client connection
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Verify Step 1/2
>>> HomeKit: [Client 4] Pair Verify Step 2/2
>>> HomeKit: [Client 4] Found pairing with 856FCE22-3DA1-4066-A021-BC1A1DC64344
>>> HomeKit: [Client 4] Verification successful, secure session established
>>> HomeKit: [Client 4] Get Accessories
>>> HomeKit: [Client 4] Remove Pairing
>>> HomeKit: Removed pairing with 856FCE22-3DA1-4066-A021-BC1A1DC64344
>>> HomeKit: Last admin pairing was removed, enabling pair setup
>>> HomeKit: Configuring mDNS
>>> HomeKit: [Client 4] Closing client connection

Detailed trace:- 

>>> HomeKit: Got IP, starting
>>> HomeKit: Configuring mDNS
>>> run_server: Staring HTTP server
>>> mdns_recv: pbuf too big
>>> HomeKit: Got new client connection: 4
>>> homekit_pairing_task: Starting pairing task
>>> homekit_pairing_task: Free heap: 29248
>>> homekit_pairing_task: [Client 4] Got 128 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 29096
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 29000
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 4] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 28336
>>> homekit_server_on_pair_setup: [Client 4] Using user-specified password: 111-11-111
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> mdns_recv: pbuf too big
>>> crypto_srp_init: Setting SRP verifier
>>> crypto_srp_get_public_key: Calculating public key
>>> send_tlv_response: [Client 4] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> tlv_debug: Type 2 value (16 bytes): \xD2\x80\xCBU\x08""\xC7\xCFk\xF0\xAD\xB3\xD1E\xB2
>>> tlv_debug: Type 3 value (384 bytes): \xDB\xA9\xF0\x92\x1BB\x1D<\xED\xF3\x88>\xFF\x90\xA7\x9F1\xF7\x11l\xF2\xDB\x97y\xC7\xF7!)|L'\xF6`\x1E_<FO)\xF0D*\x93\xEAJUH\xB9\x1F>\xAAAg\xB1uL\xE0>\xF7{\xA116\xFF\x05\xFD\xB8\xCC\x1E\x9F'\xFB|\xD3\xC4\x81\xC0\xB6\xC4\xC4\xE5\x91\x8D\x1F\x8E\xC6\xD5\x9F\xBBsk\x0F\xED\xA6\x98\x02\xB4\xD00\xDB\xF1M\xFA\xD8\xC7\\\xC6\xEE\xEFZzz\x1BF{\x86vc\xB4\x17W\xCC\xF5aMwb{\xAE\x97*\xB5\x87\xE4\xD9\xC9kG\xB9\x92\xAB\xB3U\xC9;\\\x04\xA6\x87\xB3\x9C\x87}d\xEA\xC7V]A\xE2\xCDD\xD1\xF3\xAF\xFC\xE8\x8D\x80\x92\xD0\xB9\xACS\\\xA5&F*'\xF4\x0E\xD7\x1C\x89~*\xDD\xD2u\xA6\xB7\xD0Y2\xF2\x9A\xCF_e%\xE6<\xAF\xFB\xF4\x831\xF9.\x1EH#\xD8\xEE\x9D+\x13N\x99\xF4\xFF\x02q\x05#\xC1z{\x09\x09\x82\xB9\xFA\x90\x96\x03␡\x0F\x84\x8B\xEA\xBB\xD6\xC6\x9CVOk\x07\x8E\x18\xF7\xCD&yQ^>\xFF\xCEE\xE9\xD4Vy\xEAn\x12G_\x81\x95\xCA^+\xE3\\|\x13S>`\xB1\x1B\xF4\xA1]\x06\xCC}\x8D%\xDF*{\xC00P\x84\xA3f3\x16?\xFE.\x00\xCC\xABI}:I\xC1\xBA\xCFdB\xE5cxt\x83d\xE0tq\x0F$\x8E\xFCI\x85!\xE0\x15~\x07J\xF4\xC5D\xA2m\xB9\xC4\xF4\xEE\x0AF\xC6\xA7\xA09AQ\x8Eo0\xFE\x01\xA0V\xC6\x9AT\x98bv\xFA5\xF5\xCD\x97\xB3\xCAJ\xFA\x08To@\xB2
>>> homekit_pairing_task: [Client 4] Got 256 incomming data
>>> homekit_pairing_task: [Client 4] Got 256 incomming data
>>> homekit_pairing_task: [Client 4] Got 69 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 26128
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 4 value (64 bytes): E\xDBb\xE2\xBE\x93h\x0F\x07[\xFE8\xD9\xB6\xBDPW/\x0Bl\xE9\xAB\x8FG\xC9\xA5k\xDD\x87\xCA[\x80r\xF0\x0D\xA5z\xD6t^8v\x8CC\x86\xB3\xFB\xFA\xAD\xC41\xAA\xEC\x0Dm@0\xAA\xA2\xC6\x07\x08\x9A\xA3
>>> tlv_debug: Type 3 value (384 bytes): \x1D{\x13+.\x1C\x17\x0B\xE4/\x89\x9A\x1A\xF6\xEC\xFC\xD5\xA7\xEF\x09L\xEBl\xA2.\xDFa\xEAq\xA9\xFE\xE4\x03\xC0\xAA\xA1\x18\x80t\xAF)\xE5<17\xDB\xC1\x8As\x95>\x9C5\xA5W\xBF\x8A\xB7\xFD\xAB\x00\x99\xF7J\xD9\xFF/\xB2\xC8\xDD\x0Du\xCA^\x18X\xCE\x1A\xF2\xB4\xFEx\xD9\xCC\xDC\x05f\x8B\x03]\xB7\xB3\xE2\xBD\x91T\x16`\xB5\x08q]\x8C\xA0\x94eI\x81uv{2\x8FF\xB9Um%\x96\xB0\x15\xCA\xA94\xD0\xEB\x93F\xDCK\x15\xE2\xEFI\xD0QS2\xFFg\xF8Hu&M\xE2Af\xDE\xC9\x0A\xC9\x04a\x85\xA4/\xB6|\xE0k\x0A\xD7i\x18\x87u\xBD|z-xN(\x185\x8F\xEE\x96\xE5\xC2\xBEW\x19T\x90X\xCA#\xDB\xEA\xEB\xA1\xFE\xD6\xED\xEC*\x00>\xA1\x91DB\xB2\x0Fd\xDB\xE8t!\xBFZ\xE7-WWE\xB9\x9D\xF6SpU\\\xC1\xB4\xE4%H\xC5TBl\xB4o\xA6\x8C\xA9~\x9D\x09\xBC\x93\xE7\xCF\x1Cj-\xA9\xD4\xB6\xB9\xAEv\xF1JR4x\x93\xC6\xF1\xF9\x89\x04\xC6\x8D6\xFFo\x0B\xBB\xF6\x15\xEA[\xA1\xA7-\x11D3\x08\xBF\x8D\xFDz\xB4\xE9\xEC\xAC\xD7O\xF5\xC7\xEF\x05K\xC6\xB2\x07p_\x1A&3\xE5\x04}+\x81\xD4T\x03\xE9\x85\xB25\x9E\x96\xBD\xD9?\xB1Y\xC62.}\xF4\xA8\x96␡T\xD9\x91\x0F\xDD\x1D3\x92=HmI+\x93\xEA\xB2\xE1\x1E\xEFs\x96\xD0Y\x06\xC0\x92\xCCn\x85t\xC5a\x06[\x08 \xD8\xAAHnJ\xF8\x88\xE6+U7\xAC:\x0D
>>> tlv_debug: Type 6 value (1 bytes): \x03
>>> HomeKit: [Client 4] Pair Setup Step 2/3
>>> homekit_server_on_pair_setup: Free heap: 25432
>>> homekit_server_on_pair_setup: [Client 4] Computing SRP shared secret
>>> homekit_server_on_pair_setup: Free heap: 25432
>>> mdns_recv: pbuf too big
[... 88 lines cut ...]
Content-Length: 2325
Connection: keep-alive

{"accessories":[{"aid":1,"services":[{"iid":1,"type":"3E","hidden":false,"primary":false,"characteristics":[{"aid":1,"iid":2,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"Thermostat"},{"aid":1,"iid":3,"type":"20","perms":["pr"],"description":"Manufacturer","format":"string","value":"HaPK"},{"aid":1,"iid":4,"type":"30","perms":["pr"],"description":"Serial Number","format":"string","value":"0012345"},{"aid":1,"iid":5,"type":"21","perms":["pr"],"description":"Model","format":"string","value":"MyThermostat"},{"aid":1,"iid":6,"type":"52","perms":["pr"],"description":"Firmware Revision","format":"string","value":"0.1"},{"aid":1,"iid":7,"type":"14","perms":["pw"],"description":"Identify","format":"bool"}]},{"iid":8,"type":"4A","hidden":false,"primary":true,"characteristics":[{"aid":1,"iid":9,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"Thermostat"},{"aid":1,"iid":10,"type":"F","perms":["pr","ev"],"ev":false,"description":"Current Heating Cooling State","format":"uint8","minValue":0,"maxValue":2,"minStep":1,"valid-values":[0,1,2],"value":3},{"aid":1,"iid":11,"type":"33","perms":["pr","pw","ev"],"ev":false,"description":"Target Heating Cooling State","format":"uint8","minValue":0,"maxValue":3,"minStep":1,"valid-values":[0,1,2,3],"value":3},{"aid":1,"iid":12,"type":"11","perms":["pr","ev"],"ev":false,"description":"Current Temperature","format":"float","unit":"celsius","minValue":0,"maxValue":100,"minStep":0.10000000149011612,"value":20.100000381469727},{"aid":1,"iid":13,"type":"35","perms":["pr","pw","ev"],"ev":false,"description":"Target Temperature","format":"float","unit":"celsius","minValue":10,"maxValue":38,"minStep":0.10000000149011612,"value":21},{"aid":1,"iid":14,"type":"36","perms":["pr","pw","ev"],"ev":false,"description":"Temperature Display Units","format":"uint8","minValue":0,"maxValue":1,"minStep":1,"valid-values":[0,1],"value":0}]},{"iid":15,"type":"82","hidden":false,"primary":false,"characteristics":[{"aid":1,"iid":16,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"Humidity Sensor"},{"aid":1,"iid":17,"type":"10","perms":["pr","ev"],"ev":false,"description":"Current Relative Humidity","format":"float","unit":"percentage","minValue":0,"maxValue":100,"minStep":1,"value":58.099998474121094}]}]}]}
>>> client_send: [Client 4] Encrypting payload
>>> homekit_client_task: [Client 4] Got 183 incomming data
>>> homekit_client_task: [Client 4] Decrypting data
>>> homekit_client_task: [Client 4] Decrypted 165 bytes, available 0
[... cut ...]
>>> tlv_debug: Type 0 value (1 bytes): \x04
>>> HomeKit: [Client 4] Remove Pairing
>>> HomeKit: Removed pairing with 856FCE22-3DA1-4066-A021-BC1A1DC64344
>>> HomeKit: Last admin pairing was removed, enabling pair setup
>>> HomeKit: Configuring mDNS
>>> send_tlv_response: [Client 4] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 4] Encrypting payload
>>> HomeKit: [Client 4] Force disconnect
>>> HomeKit: [Client 4] Closing client connection

maximkulkin, 2017-12-22:
This is usually an indication that you have incorrect data in your accessory definitions and controller when detects that unpairs itself automatically. I quick skim through your payload shows error in "Current Heating Cooling State" where it has value 3, although valid values are 0, 1 and 2.

maccoylton, 2017-12-22:
Thanks,that fixed it.  I have the latest code, which is a useful improvement.  BTW I notice in your example you have target temperature defaulted to 0, the range for that is normally defined as 10 to 38, so you may want to change that for the same reason that tripped me up ;-)

closed 2017-12-27 by maximkulkin

brianc118, 2018-01-14:
@maximkulkin Same issue here. I'm trying to get a garage door remote working. Where might I find specs from Apple for which characteristics I need and what they need in them? I'm currently using CURRENT_DOOR_STATE and TARGET_DOOR_STATE.

renssies, 2018-01-14:
@brianc118 you can find the specs on here, they will tell you the required characteristics, optional characteristics and their values: https://developer.apple.com/homekit/specification/. 

You do need a free developer account to access it. 

PS. if downloading redirects you to https://developer.apple.com, remove the extra slash before homekit in the URL.

michaeljanich, 2018-01-15:
I get all my info about which services use which characteristics (required or optional) from here:
https://github.com/KhaosT/HAP-NodeJS/blob/master/lib/gen/HomeKitTypes.js

maximkulkin, 2018-01-16:
@michaeljanich I think using official specification that renssies mentioned would be better

brianc118, 2018-01-16:
Yep it's working after following the official specs