maximkulkin/esp-homekit #11: sending events to self?

HomeACcessoryKid, 2017-12-23:
I observed something that might be a bug or an undesired feature (IMO).
I would say that when a specific phone sets the value, it should not receive an event of this.
Also, when setting up a new client context, I notice it happens twice in a row. Maybe that makes the event gets send to self?

In the log below, I added a 200ms repeating Task that reports heap and time...
```
--27868@0012869ms
>>> HomeKit: Got new client connection: 4
...
>>> HomeKit: [Client 4] Verification successful, secure session established
>>> homekit_client_process: [Client 4] Finished processing
--24268@0014069ms
>>> homekit_client_process: [Client 4] Got 84 incomming data
>>> homekit_client_process: [Client 4] Decrypting data
>>> homekit_client_process: [Client 4] Decrypted 66 bytes, available 0
Decrypted data (66 bytes): "GET /accessories HTTP/1.1\x0D\x0AHost: Sample\\032LED._hap._tcp.local\x0D\x0A\x0D\x0A"
>>> HomeKit: [Client 4] Get Accessories
>>> homekit_server_on_get_accessories: Free heap: 25876
>>> send_json_response: [Client 4] Sending JSON response
>>> send_json_response: [Client 4] Payload: {"accessories":[{"aid":1,"services":[{"iid":1,"type":"3E","hidden":false,"primary":false,"characteristics":[{"aid":1,"iid":2,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"Sample LED"},{"aid":1,"iid":3,"type":"20","perms":["pr"],"description":"Manufacturer","format":"string","value":"HaPK"},{"aid":1,"iid":4,"type":"30","perms":["pr"],"description":"Serial Number","format":"string","value":"037A2BABF19D"},{"aid":1,"iid":5,"type":"21","perms":["pr"],"description":"Model","format":"string","value":"MyLED"},{"aid":1,"iid":6,"type":"52","perms":["pr"],"description":"Firmware Revision","format":"string","value":"0.1"},{"aid":1,"iid":7,"type":"14","perms":["pw"],"description":"Identify","format":"bool"}]},{"iid":8,"type":"43","hidden":false,"primary":true,"characteristics":[{"aid":1,"iid":9,"type":"25","perms":["pr","pw","ev"],"ev":false,"description":"On","format":"bool","value":false}]}]}]}
>>> send_json_response: [Client 4] Sending HTTP response: HTTP/1.1 200 OK
Content-Type: application/hap+json
Content-Length: 926
Connection: keep-alive

{"accessories":[{"aid":1,"services":[{"iid":1,"type":"3E","hidden":false,"primary":false,"characteristics":[{"aid":1,"iid":2,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"Sample LED"},{"aid":1,"iid":3,"type":"20","perms":["pr"],"description":"Manufacturer","format":"string","value":"HaPK"},{"aid":1,"iid":4,"type":"30","perms":["pr"],"description":"Serial Number","format":"string","value":"037A2BABF19D"},{"aid":1,"iid":5,"type":"21","perms":["pr"],"description":"Model","format":"string","value":"MyLED"},{"aid":1,"iid":6,"type":"52","perms":["pr"],"description":"Firmware Revision","format":"string","value":"0.1"},{"aid":1,"iid":7,"type":"14","perms":["pw"],"description":"Identify","format":"bool"}]},{"iid":8,"type":"43","hidden":false,"primary":true,"characteristics":[{"aid":1,"iid":9,"type":"25","perms":["pr","pw","ev"],"ev":false,"description":"On","format":"bool","value":false}]}]}]}
>>> client_send: [Client 4] Encrypting payload
>>> homekit_client_process: [Client 4] Finished processing
--24036@0014269ms
>>> homekit_client_process: [Client 4] Got 193 incomming data
>>> homekit_client_process: [Client 4] Decrypting data
>>> homekit_client_process: [Client 4] Decrypted 175 bytes, available 0
Decrypted data (175 bytes): "PUT /characteristics HTTP/1.1\x0D\x0AHost: Sample\\032LED._hap._tcp.local\x0D\x0AContent-Length: 49\x0D\x0AContent-Type: application/hap+json\x0D\x0A\x0D\x0A{"characteristics":[{"aid":1,"iid":9,"ev":true}]}"
>>> HomeKit: [Client 4] Update Characteristics
>>> homekit_server_on_update_characteristics: [Client 4] Processing element {
	"aid":	1,
	"iid":	9,
	"ev":	true
}
>>> homekit_server_on_update_characteristics: [Client 4] There were no processing errors, sending No Content response
>>> client_send: [Client 4] Encrypting payload
>>> homekit_client_process: [Client 4] Finished processing
>>> homekit_client_process: [Client 4] Got 193 incomming data
>>> homekit_client_process: [Client 4] Decrypting data
>>> homekit_client_process: [Client 4] Decrypted 175 bytes, available 0
Decrypted data (175 bytes): "PUT /characteristics HTTP/1.1\x0D\x0AHost: Sample\\032LED._hap._tcp.local\x0D\x0AContent-Length: 49\x0D\x0AContent-Type: application/hap+json\x0D\x0A\x0D\x0A{"characteristics":[{"aid":1,"iid":9,"ev":true}]}"
>>> HomeKit: [Client 4] Update Characteristics
>>> homekit_server_on_update_characteristics: [Client 4] Processing element {
	"aid":	1,
	"iid":	9,
	"ev":	true
}
>>> homekit_server_on_update_characteristics: [Client 4] There were no processing errors, sending No Content response
>>> client_send: [Client 4] Encrypting payload
>>> homekit_client_process: [Client 4] Finished processing
--25620@0014469ms
...
--25620@0019669ms
>>> homekit_client_process: [Client 4] Got 95 incomming data
>>> homekit_client_process: [Client 4] Decrypting data
>>> homekit_client_process: [Client 4] Decrypted 77 bytes, available 0
Decrypted data (77 bytes): "GET /characteristics?id=1.9 HTTP/1.1\x0D\x0AHost: Sample\\032LED._hap._tcp.local\x0D\x0A\x0D\x0A"
>>> HomeKit: [Client 4] Get Characteristics
>>> homekit_server_on_get_characteristics: Free heap: 25564
>>> homekit_server_on_get_characteristics: [Client 4] Query paramter id = 1.9
>>> homekit_server_on_get_characteristics: [Client 4] Requested characteristic info for 1.9
>>> send_json_response: [Client 4] Sending JSON response
>>> send_json_response: [Client 4] Payload: {"characteristics":[{"aid":1,"iid":9,"value":false}]}
>>> send_json_response: [Client 4] Sending HTTP response: HTTP/1.1 200 OK
Content-Type: application/hap+json
[... 59 lines cut ...]

{"characteristics":[{"aid":1,"iid":9,"value":false}]}
>>> client_send: [Client 4] Encrypting payload
>>> homekit_server_process_notifications: [Client 4] Freeing event
--24004@0023469ms
```

report code:
```
void print_timeheap_task(void *arg) {
#ifdef HOMEKIT_DEBUG
    for(;;) {
        printf("--%05d@%07dms\n", sdk_system_get_free_heap_size(), sdk_system_get_time()/1000);
//        vTaskDelay(10);   //100ms
        vTaskDelay(20);   //200ms
//        vTaskDelay(100); //1s
    }
#endif
    vTaskDelete(NULL);
}

void homekit_server_init(homekit_server_config_t *config) {
    xTaskCreate(print_timeheap_task, "TimeHeap", 256, NULL, 1, NULL);
    xTaskCreate(homekit_server_task, "HomeKit Server", 1700, server, 1, NULL);
```

maximkulkin, 2017-12-27:
I couldn't find in documentation that sending events to controller that caused event is prohibited. I am aware that it might be a nice to have feature, but I chose implementation simplicity over slightly less network usage. Do you see any obvious drawbacks of this approach?

HomeACcessoryKid, 2017-12-27:
It was as much an attempt to show that event was armed twice. Other than that, no problem. 

BR,
HacK

maximkulkin, 2018-01-29:
Should be fixed now

closed 2018-01-29 by maximkulkin