maximkulkin/esp-homekit #13: Fatal Exception(28)

WickedGames, 2018-01-04:
Hey guys, I'm quite new to the homekit and ESP development. I've written simple led stripe controller but after a while (2-10 minutes) of seamless function my app crashes and restarts with exception (see below). Can you please at least point me in the right direction?

Fatal exception (28):
epc1=0x40105ac4

0x40105ac4: get_chunk_from_ptr at /home/src/esp/newlib-cygwin/build/xtensa-lx106-elf/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/nano-mallocr.c:202
 (inlined by) _free_r at /home/src/esp/newlib-cygwin/build/xtensa-lx106-elf/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/nano-mallocr.c:367

epc2=0x00000000
epc3=0x40203469

0x40203469: prvCheckTasksWaitingTermination at /home/zach/homekit/esp-open-rtos/FreeRTOS/Source/tasks.c:4662
 (inlined by) prvIdleTask at /home/zach/homekit/esp-open-rtos/FreeRTOS/Source/tasks.c:3241

excvaddr=0x0000000c
depc=0x00000000
excsave1=0x401059b6

0x401059b6: free at /home/src/esp/newlib-cygwin/build/xtensa-lx106-elf/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/malloc.c:173

Registers:
a0 401059b6 a1 3fff6420 a2  3fff6674 a3  00000010

0x401059b6: free at /home/src/esp/newlib-cygwin/build/xtensa-lx106-elf/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/malloc.c:173

a4  401059a4 a5  00000000 a6  3fff6020 a7  00000000

0x401059a4: free at /home/src/esp/newlib-cygwin/build/xtensa-lx106-elf/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/malloc.c:171

a8  00000003 a9  00000450 a10 00000001 a11 00000001
a12 0000000c a13 3fff6674 SAR 0000001b

Stack: SP=0x3fff6420
0x3fff6420: 3fff6628 3ffe8e80 40215320 401059b6

0x40215320: client_notify_characteristic at /home/zach/homekit/esp-homekit/src/server.c:2892

0x401059b6: free at /home/src/esp/newlib-cygwin/build/xtensa-lx106-elf/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/malloc.c:173

0x3fff6430: 00000100 3fff6674 3fff7ad4 402373ae

0x402373ae: cJSON_Delete at /home/zach/homekit/esp-homekit-demo/components/cJSON/cJSON/cJSON.c:1200

0x3fff6440: 00000000 3ffe8e80 3fff7978 3fff7978
0x3fff6450: 00000100 3ffe8e80 00000000 40237398

0x40237398: cJSON_Delete at /home/zach/homekit/esp-homekit-demo/components/cJSON/cJSON/cJSON.c:1200

0x3fff6460: 00000000 3ffe8e80 3fff7a18 3fff7a18
0x3fff6470: 00000100 3ffe8e80 00000000 40237398

0x40237398: cJSON_Delete at /home/zach/homekit/esp-homekit-demo/components/cJSON/cJSON/cJSON.c:1200

0x3fff6480: 3fff6628 00000001 3fff0978 3fff7b08
0x3fff6490: 00000100 3ffe8e80 00000000 40237398

0x40237398: cJSON_Delete at /home/zach/homekit/esp-homekit-demo/components/cJSON/cJSON/cJSON.c:1200

Free Heap: 29636
_heap_start 0x3fff1b00 brk 0x3fffa51c supervisor sp 0x40000000 sp-brk 23268 bytes
arena (total_size) 35356 fordblks (free_size) 6368 uordblocks (used_size) 28988

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2292, room 16

0x40100000: _stext at ??:?

tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4
tail 0
chksum 0x0b
csum 0x0b

maximkulkin, 2018-01-04:
It looks like you're using some old version of esp-homekit. You need to update your version and rebuild firmware.

WickedGames, 2018-01-04:
I have a clone of this repository and `git pull` says it's up to date. I even did `git submodule update --recursive` but nothing. Am I doing something wrong?

carmanuelz, 2018-01-05:
I have de same error:

Fatal exception (28):
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000

I did git submodule update --recursive.

maccoylton, 2018-01-06:
Make sure you pull the every latest version of both esp-open-rtos  and esp-HomeKit, both have improvements in memory management. 

From what I’ve seen above message tends to be running out of heap space or your code trying to access a piece of memory it shouldn’t.  

If you are running one of the demo’s unmodified then it will likely be the former.

BTW what chip are you compiling for ?

carmanuelz, 2018-01-06:
I'm using ESP-8266 on this devkit
[image]

maccoylton, 2018-01-06:
That model only has 1M of Flash according to this:- https://www.aliexpress.com/store/product/Official-DOIT-New-Version-1pcs-ESP-12N-Serial-Wireless-WiFi-Transmission-Module-Fully-Compatible-with-ESP8266/636052_32812237494.html

To build for this, you need to define

WickedGames, 2018-01-06:
Ok, great. I think that we are getting somewhere with the flash size. Is it somehow possible to use some tool (probably esptool.py) to get flash size of currently connected chip? Sorry not to google it my self. But I'm on vacation and internet connection sucks here. But at the same time I don't want this discussion to die :)

carmanuelz, 2018-01-06:
I solved the problem by adding a flag in the make command
`make flash FLASH_MODE=dout`
now it works correctly.
the model of my devkit is nodemcu 1.0 (esp-12e)
This model has flash size 4M (1M SPIFFS)

closed 2018-01-09 by maximkulkin