maximkulkin/esp-homekit #16: How to add Hue Caracteristic

carmanuelz, 2018-01-08:
Hi,
I try to add Hue characteristic like this way, but doesn't work
`
            HOMEKIT_CHARACTERISTIC(
                HUE, 100,
                .getter=led_hue_get,
                .setter=led_hue_set
            ),`

Could you help me with this?

maximkulkin, 2018-01-08:
I will try to read mind here: by "it doesn't work" you mean that you do not see Hue controls on iOS UI. If so, thing is that you need to also have SATURATION characteristic for it to work with iOS.

HomeACcessoryKid, 2018-01-08:
see my ZemiSmart example added (it is a pull request for now) to the esp-homekit-demo. It works all the way (see my youtube channel)

carmanuelz, 2018-01-09:
>I will try to read mind here: by "it doesn't work" you mean that you do not see Hue controls on iOS UI. If so, thing is that you need to also have SATURATION characteristic for it to work with iOS.
@maximkulkin You are right, It was my mistake to check the HAP documentation
@HomeACcessoryKid Thank you very much for the example.

closed 2018-01-09 by carmanuelz