maximkulkin/esp-homekit #20: posix implementation

aep, 2018-01-23:
Hi,

we're considering using this implementation on a linux based system. Would you generally be opposed to adding a whole bunch of ifdefs for posix compatibility?

maximkulkin, 2018-01-23:
If you have a luxury of running a Linux based system, you can use a much more decent and Linux-ready implementations, e.g. https://github.com/KhaosT/HAP-NodeJS or https://github.com/brutella/hc.

aep, 2018-01-23:
Not desktop Linux, embedded mips (no golang) with 4mb flash (no gnome libs or JavaScript).

We're almost done porting. But it's quite ugly due to the different threading stuff. Have a look at https://github.com/korhalio/esp-homekit at let me know if you'd still take it. Well clean it up then.

aep, 2018-01-24:
@maximkulkin we're done i think but my test iTHing doesnt accept the characteristics. any idea where I broke it?

```
>>> homekit_client_process: [Client 6] Got 94 incomming data
>>> homekit_client_process: [Client 6] Decrypting data
>>> homekit_client_process: [Client 6] Decrypted 76 bytes, available 0
>>> HomeKit: [Client 6] Get Accessories
>>> client_send: [Client 6] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/hap+json\x0D\x0ATransfer-Encoding: chunked\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A
>>> client_send: [Client 6] Sending payload: 3fb\x0D\x0A{"accessories":[{"aid":1,"services":[{"iid":1,"type":"3E","hidden":false,"primary":false,"characteristics":[{"aid":1,"iid":2,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"Sonoff Switch-040506"},{"aid":1,"iid":3,"type":"20","perms":["pr"],"description":"Manufacturer","format":"string","value":"iTEAD"},{"aid":1,"iid":4,"type":"30","perms":["pr"],"description":"Serial Number","format":"string","value":"037A2BABF19D"},{"aid":1,"iid":5,"type":"21","perms":["pr"],"description":"Model","format":"string","value":"Basic"},{"aid":1,"iid":6,"type":"52","perms":["pr"],"description":"Firmware Revision","format":"string","value":"0.1"},{"aid":1,"iid":7,"type":"14","perms":["pw"],"description":"Identify","format":"bool"}]},{"iid":8,"type":"49","hidden":false,"primary":true,"characteristics":[{"aid":1,"iid":9,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"Sonoff Switch"},{"aid":1,"iid":10,"type":"25","perms":["pr","pw","ev"],"ev":false,"description":"On","format":\x0D\x0A
>>> client_send: [Client 6] Sending payload: 1d\x0D\x0A"@\x19'\x17\x10V","value":false}]}]}]}\x0D\x0A
>>> client_send: [Client 6] Sending payload: 0\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 6] Finished processing
>>> HomeKit: [Client 6] Closing client connection

```

iphone says "could not connect"

maximkulkin, 2018-01-24:
It looks like there was a bug in json library handling var args on buffer boundaries. I just pushed a patch. Please update and see if it fixes.

aep, 2018-01-25:
thanks! we got it working on x86 but not mips. this the output on the mips machine:
```

>>> client_send: [Client 6] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/hap+json\x0D\x0ATransfer-Encoding: chunked\x0D\x0AConnecti
>>> client_send: [Client 6] Sending payload: 404\x0D\x0A{"accessories":[{"aid":0,"services":[{"iid":0,"type":"3E","hidden":false,"primary":false,"characteristics":[{"aid":0,"iid":0,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"Switch-040506"},{"aid":0,"iid":0,"type":"20","perms":["pr"],"description":"Manufacturer","format":"string","value":"kh"},{"aid":0,"iid":0,"type":"30","perms":["pr"],"description":"Serial Number","format":"string","value":"037A2BABF19D"},{"aid":0,"iid":0,"type":"21","perms":["pr"],"description":"Model","format":"string","value":"Basic"},{"aid":0,"iid":0,"type":"52","perms":["pr"],"description":"Firmware Revision","format":"string","value":"0.1"},{"aid":0,"iid":0,"type":"14","perms":["pw"],"description":"Identify","format":"bool"}]},{"iid":0,"type":"49","hidden":false,"primary":true,"characteristics":[{"aid":0,"iid":0,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"Switch"},{"aid":0,"iid":0,"type":"25","perms":["pr","pw","ev"],"ev":false,"description":"On","format":"bool","value":false}]}
>>> client_send: [Client 6] Sending payload: 0\x0D
>>> homekit_client_process: [Client 6] Finished processing
>>> homekit_run_server: .
>>> homekit_run_server: client read
>>> homekit_client_process: [Client 6] Got 186 incomming data
>>> homekit_client_process: [Client 6] Decrypting data
>>> homekit_client_process: [Client 6] Decrypted 168 bytes, available 0
>>> homekit_server_on_pairings: HomeKit Pairings
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 1 value (36 bytes): 458964B5-9CFC-4E4A-848D-838022FF525E
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> tlv_debug: Type 0 value (1 bytes): \x04
>>> HomeKit: [Client 6] Remove Pairing
>>> HomeKit: Removed pairing with 458964B5-9CFC-4E4A-848D-838022FF525E
>>> HomeKit: Last admin pairing was removed, enabling pair setup
```

aep, 2018-01-25:
got it. it must be json_write(json, "%lld", x); in  json_integer

maximkulkin, 2018-01-25:
Not sure those are related, but definitely I see all your iid attributes being zero.

aep, 2018-01-25:
yes it is. using %lld instead of %ld fixed it.
https://twitter.com/arvidep/status/956611266985934848

user623523, 2018-03-20:
This is awesome, I love having a posix implementation!
This means I can integrate HomeKit with my home automation system, without relying on nodejs based stuff.
I added another file descriptor in server.c to communicate events from my home automation system into esp-homekit, and that works fine.

Anyway, thanks guys, keep up the good work!

aep, 2018-03-20:
Cool! But we're not maintaining this fork since it's rather messy. Instead we started implementing the whole thing in rust .  https://github.com/korhalio/hap-rs

user623523, 2018-03-21:
I'm sorry to hear that, rust is a no go for me.

maximkulkin: Let me know if you need a bit of development resources put into this, I'll do what I can.

maximkulkin, 2018-03-22:
@user623523 Just to let you know, I do not plan to port it to POSIX in the near future. Currently I'm working on ESP32 port and maybe later I will look into Arduino port (where ESP8266 is used as WiFi module), but no other plans so far.

younghyunjo, 2018-04-02:
@maximkulkin 
I am done with ESP32.
Please visit my repository.
https://github.com/younghyunjo/esp32-homekit