maximkulkin/esp-homekit #30: Function to rebroadcast accessory and/or verify if already connected to a client

Arduingo, 2018-02-14:
In some situations when the accessory is first power up clients like the Apple TV or the iOS Home App will not connect immediately (HomeApp connect after open it), this limits the ability for the accessory to send events on power up.

From the logs I have taken I was able to find that after some time there seems to be an incoming client connection and after that notifications and automations start working properly (Have not yet measured if this time is constant)

I have not been able to find any public functions to rebroadcast accessories or verify if it's already connected in order to hold the event broadcast until the accessory is really connected to a client, without this power up event get lost.

maximkulkin, 2018-02-16:
This is not in the nature of accessories. If accessories have no clients, events are lost. For a really important events there are special means to retrospect them (e.g. Logs characteristic for Lock Management service). Other than that it is assumed that 1) simple controllers like iPhones do not need to be constantly connected to accessories; 2) controllers like iPad, Apple TV or HomePod (that are configured as automation servers) should immediately connect once they see accessory mDNS advertisement.

Arduingo, 2018-02-16:
Hi, thanks for your answer

Its the option 2 witch is failing if the accessory in power cycled and stay offline for less than 2 min. if the mDNS record have not been removed the new announce will not get effect, the notification or update get lost and the automations depending on them aren't triggered.

in My test I monitored with dns-sd -B -hap local, and found that if the record is still alive the ATV will not connect to the accessory when it announces upon power up.

If before power up again the accessory I wait for the RMV of the record, the ATV connect inmediatly as expected. This could be important in sensors used in home security, 

One question is the current mDNS implementation taking care of name collisions?

maximkulkin, 2018-02-16:
No, it does not take care of name collisions. However, there is a technique to minimize those by generating accessory names based on mac addresses (see [this example](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/accessory_names/main.c) and [this](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/sonoff_basic/main.c#L176-L187)).

I will test how (and if) official accessories handle name collisions.

closed 2019-01-28 by maximkulkin