maximkulkin/esp-homekit #31: mdns_recv: pbuf too big

bobmosh, 2018-02-16:
Hey,

I followed the led example from [here,](https://github.com/maximkulkin/esp-homekit-demo) and I ran it properly the first time. However, I didn't change the code, (I even pulled the repo completely new) and now I'm getting an error Message when I monitor the ESP via terminal:

```
Opening /dev/cu.SLAB_USBtoUART at 115200bps...
pp_task_hdl : 3ffefd90, prio:14, stack:512
pm_task_hdl : 3ffef7f0, prio:1, stack:176
frc2_timer_task_hdl:0x3ffeeac0, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Feb 16 2018 13:12:41
phy ver: 273, pp ver: 8.3

mode : sta(5c:cf:7f:76:f8:1a)
add if0
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: A7:24:08:04:91:30
>>> HomeKit: Found admin pairing with 2A4EABB8-9692-4D05-B603-D53899BAD14E, disabling pair setup
>>> HomeKit: Waiting for IP
scandone
add 0
aid 3
cnt

connected with G&W-Lan, channel 6
dhcp client start...
ip:192.168.0.24,mask:255.255.255.0,gw:192.168.0.1
>>> HomeKit: Got IP, starting
>>> HomeKit: Configuring mDNS
>>> mdns_recv: pbuf too big
>>> mdns_recv: pbuf too big
```

Any idea why this is and how I can resolve it?
It ran properly today, but now it doesn't anymore.

renssies, 2018-02-16:
`mdns_recv: pbuf too big` is more like a warning. Not really an error. 

However, I believe the esp-homekit submodule was updated to address this. Maybe try setting the esp-homekit submodule to its master

bobmosh, 2018-02-16:
Thanks for the reply! I'll give that a shot!

However I'm not even able to add the led example to my HomeApp.
Seems to be a different issue though. I'll keep trying for now.

hn-nguyen, 2018-02-17:
Did you try to clean your esp by using `Make erase_flash` ? 

I guess that you removed an accessory from HomeApp when its esp device is off. Thus when you take the esp on, it does not restart a pairing as it found a old key.

bobmosh, 2018-02-17:
@hn-nguyen Actually, I didn't think of that. I was assuming just overwriting would be enough...
I just tested it though! It works! Thanks so much!

hn-nguyen, 2018-02-18:
You're welcome.

closed 2018-02-23 by maximkulkin