maximkulkin/esp-homekit #34: accessory not discoverable right after wifi configuration and connection to AP

ageorgios, 2018-02-28:
Tried using wifi_config.h
but when it connects to the AP the homekit device is not discoverable and it needs a reset.

```
void on_wifi_ready() {
    homekit_server_init(&config);
}

void user_init(void) {
    uart_set_baud(0, 115200);
    create_accessory_name();
    gpio_init();
    wifi_config_init("ESP-Homekit-Motor", NULL, on_wifi_ready);
}
```

maximkulkin, 2018-02-28:
logs please

ageorgios, 2018-03-01:
This is a case where ERASE->FLASH->CONNECT AP->PAIR(fails)

```
Booting rom 0.
pp_task_hdl : 3ffefdd0, prio:14, stack:512
pm_task_hdl : 3ffef830, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3de0, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Feb 28 2018 21:43:08
phy ver: 273, pp ver: 8.3

>>> wifi_config: wifi_config_station_connect: No configuration found
>>> wifi_config: Starting AP mode
>>> wifi_config: wifi_config_softap_start: Starting AP SSID=ESP-Homekit-Motor-121C4C
>>> wifi_config: Starting DHCP server
mode : sta(a0:20:a6:12:1c:4c) + softAP(a2:20:a6:12:1c:4c)
add if0
add if1
bcn 100
>>> wifi_config: Starting WiFi scan
>>> wifi_config: Starting DNS server
>>> wifi_config: Staring HTTP server
scandone
scandone
add 1
aid 1
station: 04:54:53:07:3d:db join, AID = 1
>>> wifi_config: http_task: Got 64 incomming data
!!! wifi_config: Unknown endpoint: GET /hotspot-detect.html
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 3 incomming data
>>> wifi_config: wifi_config_server_on_message_complete: Unknown endpoint
>>> wifi_config: client_send_redirect: Redirecting to http://192.168.4.1/settings
scandone
>>> wifi_config: Client disconnected
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 50 incomming data
>>> wifi_config: Client disconnected
>>> wifi_config: http_task: Got 64 incomming data
!!! wifi_config: Unknown endpoint: GET /hotspot-detect.html
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 26 incomming data
>>> wifi_config: wifi_config_server_on_message_complete: Unknown endpoint
>>> wifi_config: client_send_redirect: Redirecting to http://192.168.4.1/settings
>>> wifi_config: Client disconnected
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 9 incomming data
>>> wifi_config: Client disconnected
>>> wifi_config: http_task: Got 64 incomming data
!!! wifi_config: Unknown endpoint: GET /hotspot-detect.html
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 3 incomming data
>>> wifi_config: wifi_config_server_on_message_complete: Unknown endpoint
>>> wifi_config: client_send_redirect: Redirecting to http://192.168.4.1/settings
>>> wifi_config: Client disconnected
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 50 incomming data
>>> wifi_config: Client disconnected
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 64 incomming data
>>> wifi_config: http_task: Got 64 incomming data
[... 86 lines cut ...]
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> HomeKit: [Client 6] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 17576
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 6] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 16912
>>> homekit_server_on_pair_setup: [Client 6] Using user-specified password: 111-11-111
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Failed to get SRP verifier (code -2)
>>> crypto_srp_get_public_key: Calculating public key
!!! HomeKit: [Client 6] Failed to dump SPR public key (code -2)
>>> send_tlv_response: [Client 6] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 7 value (1 bytes): \x01
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 6] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 6\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x07\x01\x01\x06\x01\x02
>>> homekit_client_process: [Client 6] Finished processing
>>> HomeKit: [Client 6] Closing client connection
```

maximkulkin, 2018-03-01:
> !!! HomeKit: [Client 6] Failed to dump SPR public key (code -2)

Code "-2" means "out of memory".

> \>\>\> HomeKit: [Client 6] Pair Setup Step 1/3
> \>\>\> homekit_server_on_pair_setup: Free heap: 17576

At this point you should have at least ~23K of RAM, otherwise crypto will fail.

This could be related to memory leak in wifi_config which I'm working on now, but also it could be related to stuff that YOU allocate (e.g. your own tasks). Usually wifi_config consumes ~15K which leaves RAM down to ~27K which should be sufficient for pairing.

ageorgios, 2018-03-01:
Thanks for your clarifications,

the init proc is above, nothing memory intensive

maximkulkin, 2018-03-01:
Then I suggest you to reset accessory after configuring WiFi before pairing until I fix memory usage in wifi_config.

ageorgios, 2018-03-01:
Thanks for the great work!

closed 2018-03-01 by ageorgios