maximkulkin/esp-homekit #36: A callback when accessoire is ready

hn-nguyen, 2018-04-03:
I wonder if there exists some `callback` to let user know when the accessoire is ready, such as pairing process has been successfully done, ..

This callback is really helpful to initialize other tasks, for example, creating a task to read PIR status, ... These tasks cannot be started at beginning as they will take memory, then, there exists very less memory for homekit paring.

maximkulkin, 2018-04-03:
What will you do if user will unpair device? Will your accessory stop doing its primary function until another pairing?

nhnghia, 2018-04-03:
Yes, that is a good idea as it’s inutile working without notification to user. In such a case I think that a callback when unpairing is also wanted.

maximkulkin, 2019-02-02:
You should be able to track some internal state of homekit through a new event notifications. Check out [led_status example](https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/led_status).

closed 2019-02-02 by maximkulkin