maximkulkin/esp-homekit #44: QR code

korstiaanS, 2018-04-24:
Hi,

I want to generate QR codes for the devices I make.
I could use the Homekit simulator to fill in all the exact values and this program will generate the QR code I could use.
In the ESP HomeKit code I find all the necessary fields (Name, Manufacturer, Serial, Model, Password, ...) except something for the SETUP ID. I suppose I need this setup ID field in the ESP as well.
Is that field available and/or can this be added?

Korstiaan

korstiaanS, 2018-04-24:
This is the simulator:

[image]

renssies, 2018-04-24:
Using the simulator wouldn't work if you knew the Setup ID. The code needs to generate a certain hash that is broadcasted, this is currently not in the code. 

@cduvenhorst was working on implementing it into esp-homekit: https://github.com/maximkulkin/esp-homekit/pull/33

korstiaanS, 2018-04-24:
Ah, ok. Sorry. I didn't knew it was that complicated.
Thanks for the answer.
It was a nice thing to have (and show off)...

royfocker19, 2018-07-05:
@korstiaanS, what'sthe name of that simulator.? Where can I download it?

korstiaanS, 2018-07-06:
@royfocker19 : it is called HomeKit Accessory Simulator. It is from Apple and can be downloaded from the Apple Developer Site. See [here](https://stackoverflow.com/questions/49004759/how-to-use-home-kit-accessory-simulator)

fmoreni, 2018-09-09:
Hello, my friend,

I did not find on the apple website the qrcode simulator, did you get it where?

Could you help please?

cduvenhorst, 2018-09-09:
Since Xcode 8 the HomeKit Accessory Simulator is provided with “Additional Tools for Xcode”. https://developer.apple.com/download/more/?=for%20Xcode

AramVartanyan, 2018-10-13:
You can also use this way:

[HomeKit Codes.zip](https://github.com/maximkulkin/esp-homekit/files/2475344/HomeKit.Codes.zip)

It works for me.

maximkulkin, 2019-02-17:
Added support for QR code pairing, see [QR code pairing](https://github.com/maximkulkin/esp-homekit/tree/master#qr-code-pairing).

closed 2019-02-17 by maximkulkin