maximkulkin/esp-homekit #45: Silence the serial port

cduvenhorst, 2018-05-03:
Is there a way to silence the outputs on the serial port?

maximkulkin, 2018-05-09:
Well, as with other parts of SDK which unconditionally write to serial console, I guess the only option is to not initialize UART (uart_set_baud()).

cduvenhorst, 2018-05-10:
Well, that‘s what Is quite obvious but - take the Sonoff SC for instance, which communicates with the sensor unit via the UART with AT+ commands. Currently the homekit-info messages all get printed to the sensor unit which reads the UART.

I took a first stab to solve this little problem with the definition of debug levels (INFO, WARN, DEBUG, ...) which works for me now. I am not quite sure if this is of wide interest, since this seems to be a special case.

maximkulkin, 2018-05-10:
Thing is that as I said apart from esp-homekit itself there are quite a few places inside SDK that also write to serial (e.g. when you connect to WiFI it prints it's state). I've seen that people who really want to silence everything use some linker magic to substitute `printf()` function with a dummy one.

Just a small hint: https://github.com/SuperHouse/esp-open-rtos/blob/master/common.mk#L209-L211

closed 2018-05-10 by maximkulkin