maximkulkin/esp-homekit #47: Android integration

tomasrocchetti, 2018-06-19:
I think this project is amazing, but anyone tried to add some libraries to make the software compatible with other domotic systems like domoticz, home assistant or Google home?

RavenSystem, 2018-06-19:
This project is for Apple HomeKit, and IMO I think that anybody who uses HomeKit doesn't want to combine it with other systems; but you can find some projects using other systems for ESP chips simply searching into GitHub, for example: https://github.com/search?q=esp+domoticz

gongloo, 2018-06-19:
While I do believe that adding additional compatibility layers inside this framework as it stands today is outside the scope of this project, I disagree with your assessment that "anybody who uses HomeKit doesn't want to combine it with other systems". There are plenty homes with mixed ecosystems, not least families who mix Android and iPhone products (I.E. one partner uses an iPhone, another uses Android). Without a doubt there are literally millions of people with Google Home devices who also use iOS devices daily.

tomasrocchetti, 2018-06-19:
I agree with you, do you know any repositories that allow me to create compatible devices with google home (or domoticz) using FreeRTOS?
In this way I can try to use both ecosystems.
Thank you in advance

maximkulkin, 2019-01-28:
@gongloo You could just put Android frameworks next to HomeKit (provided those also work on FreeRTOS). The problem is: ESP8266 is too limited on resources to put multiple frameworks there. I think in mixed setup your best bet is to use bridge software (like [Homebridge](https://github.com/nfarina/homebridge)) installed on some linux system (e.g. on Raspberry Pi).

closed 2019-01-28 by maximkulkin