maximkulkin/esp-homekit #48: Answer to list pairings seems to be reversed

jlusiardi, 2018-06-29:
Hi,

the answer to list pairings (server.c lines 2625ff) seems to be reversed. IMHO list pairings is one of the few occasions that TLVs are dependend on a special order. The spec on page 55ff says to construct a TLV with items:
- kTLVType_State <M2>
- kTLVType_Idendtifier <PairingID controller 1>
- kTLVType_PublicKey <Public key controller 1>
- kTLVType_Permissions <Permissions controller 1>
- kTLVType_Separator
- kTLVType_Idendtifier <PairingID controller ...>
- kTLVType_PublicKey <Public key controller ...>
- kTLVType_Permissions <Permissions controller ...>

Everything after the separator is repeated for all additional pairings.

If I do such a request to a esp-homekit on a esp8266 I get:
In python:
```
[(11, bytearray(b'\x01')), (3, bytearray(b'Q<cWH\x05*\xe1\x13`\xd1\x11\xb1\xb4\xa2\x9d\xda7\xb84+w\xc6\xa59\x04\x993\xb07\xcd\xc3')), (1, bytearray(b'764e5722-07e6-464b-82f6-3d2c17f2185f')), (6, bytearray(b'\x02'))]
```
and also while monitoring the esp over serial line:
```
>>> HomeKit: [Client 4] List Pairings
>>> send_tlv_response: [Client 4] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 11 value (1 bytes): \x01
>>> tlv_debug: Type 3 value (32 bytes): Q<cWH\x05*\xE1\x13`\xD1\x11\xB1\xB4\xA2\x9D\xDA7\xB84+w\xC6\xA59\x04\x993\xB07\xCD\xC3
>>> tlv_debug: Type 1 value (36 bytes): 764e5722-07e6-464b-82f6-3d2c17f2185f
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 4] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 78\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x0B\x01\x01\x03 Q<cWH\x05*\xE1\x13`\xD1\x11\xB1\xB4\xA2\x9D\xDA7\xB84+w\xC6\xA59\x04\x993\xB07\xCD\xC3\x01$764e5722-07e6-464b-82f6-3d2c17f2185f\x06\x01\x02
>>> homekit_client_process: [Client 4] Finished processing
>>> HomeKit: [Client 4] Closing client connection
```

Is it possible to reverse the order of the items in the TLV so it is according to the specs? This would be a big advantage!

Regards
Joachim

maximkulkin, 2018-08-09:
The spec says that "The response is a group of TLV items with separator items to delineate them. Each pairing entry must be comprised of the following TLV items". It does not mention that those TLV items should be in exact order. Although this was not hard to change.

closed 2018-08-09 by maximkulkin

jlusiardi, 2018-08-10:
Hi Maxim,

i'll try to test this this weekend. Concerning the specs, i also thought, that ordering was irrelevant. But as far as I have understood, the spec never says that the order may be changed at will.

Regards
Joachim