maximkulkin/esp-homekit #49: Integration with MQTT

pburgio, 2018-07-04:
Hello,

I am trying to integrate the HK server to make it work side-by-side with an MQTT client. Unfortunately, when I run both the modules, the HK server gets stuck (doesn't even start, at least, it seems so to me..)

Did anyone ever tried to do so, or to make it work, say, with some kind other "client module", such as for instance HTTP?

Best
Paolo

user623523, 2018-07-24:
I did an integration with my home automation system, which in turn uses MQTT.

I'm sending text notifications from my HA system to esp-homekit, so I had to add an additional socket that esp-homekit monitors.
Depending on how your MQTT library works you need to drive both HK and MQTT, so you can't have one block the other.

I've used the lwmqtt library and it can be used in a non blocking fashion, so perhaps you could look into that.

fanlessfan, 2018-09-04:
Could we have an MQTT example in esp-homekit-demo?

fanlessfan, 2018-09-07:
@user623523  Could you share your sample code with non blocking fashion? thx

maximkulkin, 2019-01-28:
There are no plans for a MQTT integration

closed 2019-01-28 by maximkulkin