maximkulkin/esp-homekit #52: implicit declaration of function 'HOMEKIT_DECLARE_CHARACTERISTIC_SELECTED_RTPS_CONFIGURATION'

jonkofee, 2018-07-23:
How setup camera?

Please help

maximkulkin, 2018-07-24:
Well, I guess nobody ever tried to set up camera with this framework, because I guess there is no way to connect camera and/or enough resources to stream video through ESP8266.

Why do you need this?

jonkofee, 2018-07-24:
I want watching my door

maximkulkin, 2018-07-24:
Ok, I guess you know what you're doing and the only issue you have is with missing characteristic declaration which I can help you with. So, according to Homekit Accessory Protocol specification the name of characteristic you're looking for is "SELECTED_RTP_STREAM_CONFIGURATION" (not "SELECTED_RTPS_CONFIGURATION"). The comment in our header file is incorrect and I will update it.

closed 2018-08-03 by jonkofee