maximkulkin/esp-homekit #62: Homekit Speaker

AustinRobinson, 2018-10-15:
Are there any examples or can someone provide an example on how to get a speaker working with this library?

renssies, 2018-10-15:
Making a speaker with this library in impossible. A HomeKit speaker is a speaker with WiFi and AirPlay 2. And sadly AirPlay 2 is limited to MFi licensees who have a strict NDA.

However people could reverse engineer airplay 2 (like the did with airplay 1) but unfortunately that hasn’t happened yet

AustinRobinson, 2018-10-15:
Is there any documentation on how they reverse engineered airplay 1?

AdySan, 2018-10-15:
See this https://github.com/mikebrady/shairport-sync/issues/535

timoschilling, 2018-10-15:
HomeKit Speaker is not the same as an AirPlay Speaker.

It is right what @renssies says About the AirPlay 2 Speeker.

But it is possible to build an HomeKit Speaker with this lib.

Take a look Accessory Identifier and the related Service and add them to this projekt.

Here are the related files from an Go based lib:
https://github.com/brutella/hc/blob/master/service/speaker.go
https://github.com/brutella/hc/blob/master/characteristic/mute.go
https://github.com/brutella/hc/blob/master/characteristic/volume.go

renssies, 2018-10-15:
@timoschilling: those are the speaker services that can only used for cameras and video doorbells sadly

timoschilling, 2018-10-15:
That's not true!

Here is an Devise Created by the original Apple HomeKit Accessory Simulator:
[image]
[image]

renssies, 2018-10-16:
Yeah, that is speaker control, it is made for controlling the speaker of cameras and video doorbells. It's not supported by the official Home app but you can use it to build your own accessories to control a speaker using a different HomeKit (like the Home app by Matthias that @timoschilling is using)

However it won't let you play music to it. That is only for AirPlay 2.

timoschilling, 2018-10-16:
Sure, thats HomeKit works! But your statement `Making a speaker with this library in impossible.` isn't right, it's possible. HomeKit this for controlling things not for streaming Media.

timoschilling, 2018-10-16:
It's like building an doorbell or an webcam with HomeKit, the controlling is the domain of HomeKit the Streaming technology not.

AustinRobinson, 2018-10-16:
So to be able to stream music, Airplay 2 would need to be reverse engineered?

renssies, 2018-10-16:
That is true, however, most people that mean a "HomeKit speaker" mean a speaker that appears in the Home app, which is an AirPlay 2 speaker. 

So yes, you can build a HomeKit speaker with this library, meaning a speaker that you can control with a third party home app. However, you can't build a speaker that you can control using the Home app or stream music to. 

And yes, in order to make a speaker that appears in the Home app and that you can stream music too from throughout iOS it will need to be an AirPlay 2 speaker. And in order to make one as a maker/hobbyist, you will have to wait for AirPlay 2 to be reverse engineered :)

AustinRobinson, 2018-10-16:
Ok thanks

timoschilling, 2018-10-16:
There is one other DIY way, buy an Apple AirPort Express A1392 wich has an 3.5 mm audiojack and build it in your speaker. The A1392 has become AirPlay 2 support via an software update.

mentaluproar, 2019-06-25:
Will this do what you want?
https://github.com/mikebrady/shairport-sync

maximkulkin, 2019-06-25:
@mentaluproar I am sure that there are numerous programs that implement that. The idea here was to put everything onto a $5 hardware.

mentaluproar, 2019-12-08:
> Any news?

Airplay2 was recently reverse engineered. I’d expect renewed interest in this soon.

maximkulkin, 2019-12-08:
I would say, interest would renew if there would be off the shelf kits with ESP32 and speaker.

maximkulkin, 2020-01-27:
Closing this for now, not gonna happen in near future

closed 2020-01-27 by maximkulkin

maximkulkin, 2020-01-27:
Because do not plan to do that

Neustradamus, 2020-01-27:
It is not good to close this ticket if not solved... One guy can develop it...

maximkulkin, 2020-01-27:
1. Didn't happen so far. Frankly, nobody here ever worked on an issue (he did not report himself), so this is list is mostly for me and I do not want it hanging there.
2. This issue is not related to HomeKit in general, AirPlay has nothing to do with HomeKit. Issue list is not a collection of random notes and memes from internets, it's a list of what needs to be done for this particular repo.