maximkulkin/esp-homekit #63: homekit_characteristic_t Not Changing

AustinRobinson, 2018-10-18:
If I call `homekit_characteristic_t current = HOMEKIT_CHARACTERISTIC_(CURRENT_DOOR_STATE, 0);` and then call `printf("Current: %d\n", current.value.int_value);` it outputs "Current: 0" as expected. But when I call `homekit_characteristic_notify(&current, HOMEKIT_UINT8(3));` it still outputs "Current: 0"

renssies, 2018-10-18:
See “button_callback” here: https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/sonoff_basic/main.c

I set the value of the characteristic first, then I send the notification.

AustinRobinson, 2018-10-18:
Never mind, I figured it out. have to do `current.value = HOMEKIT_UINT8(3);` and then call `homekit_characteristic_notify(&current, current.value);` Thanks for your help.

AustinRobinson, 2018-10-19:
Even though current.value is 1, and my phone gets the "Default Room Garage Door Was Closed" it says opening in the Home App.

renssies, 2018-10-19:
The home app keeps saying opening... or closing... until obstruction detected is false and the current position matches the target position. So check the value of target door state.

AustinRobinson, 2018-10-19:
Ok I get it now, thanks.

closed 2018-10-19 by AustinRobinson