maximkulkin/esp-homekit #64: Irrigation category

conkerkh, 2018-10-20:
I can't seem to find specs for category of irrigation system. Several services include them however I have no idea for example what states can be defined in Program Mode etc, seems like official Apple documentation lacks any specs for sprinkler, faucet and irrigation system. Is there any source I can look at to get an idea how this works?

AramVartanyan, 2018-10-20:
I am working on this problem right now. And I am having a progress.
So far I found out that it is better to use VALVE, instead of IRRIGATION_SYSTTEM.
Also to make it work, you should change some rows into types.h.

I will upload my model soon. 

About the characteristics details Apple says that the irrigation, valve, faucet etc are for the certified MFi only.
However you can find more details here:

https://github.com/Colorado4Wheeler/HomeKit-Bridge/wiki/HomeKit-Model-Reference

RavenSystem, 2018-10-20:
My firmware manages a water valve for irrigation systems (device type 7):
https://github.com/RavenSystem/esp-homekit-devices/blob/master/devices/Sonoff_RavenCore/main.c

maximkulkin, 2019-01-28:
Valve, Irrigation System and Faucet services can be found in homekit/characteristics.h

closed 2019-01-28 by maximkulkin