maximkulkin/esp-homekit #70: Display status information

GermanSheepDog, 2019-01-27:
Hello,
I need some status information for my OLED-display:
- Is homekit paired?
- Is homekit pairing?
- Get the current (and not the last) accessory_id?

I like functions like this:
```
bool homekit_server_paired();
bool homekit_server_pairing();
char* homekit_server_accessory_id();
```

Is it possible to get the new procedures?

peros550, 2019-01-28:
@RavenSystem @GermanSheepDog 

In regards to the pairing status, I have made a proposal in this issue. You may check is the proposed code changes are good enough or not and comment. 

https://github.com/maximkulkin/esp-homekit/issues/54

peros550, 2019-01-28:
> @peros550
> 
> * Every time a updated github I lost my changes.

I have the same issue. Wondering how could that be overcome if Maxim decides not to implement the proposed changes in server.c file

maximkulkin, 2019-02-02:
I have added support for events to esp-homekit. Currently these events are supported:
* server initialized
* client connected
* client verified
* client disconnected
* pairing added
* pairing removed

Also, I've added function to get paired status (if there are any pairings) and function to query accessory ID.

closed 2019-02-02 by maximkulkin

maximkulkin, 2019-02-02:
No, there is no callback for WiFi, since HomeKit framework does not manage WiFi. On ESP32 ESP-IDF SDK allows you to specify callback for WiFi changes, but on ESP_OPEN_RTOS you need to poll WiFi status manually.

Also, some HomeKit examples use esp-wifi-config library for which I also plan to add support for events (similar to ESP-IDF).

peros550, 2019-02-02:
thanks for the feedback! Have a nice day!

GermanSheepDog, 2019-02-03:
@maximkulkin: Many tanks for this new home kit functions. The new functions worked great in my environment.

I found a little challenge at led_status example (line 6):
```
void on_event(homekit_event_t event) {
    if (event == HOMEKIT_EVENT_SERVER_INITIALIZED) {
        led_status_set(led_status, paired ? &normal_mode : &unpaired);
    }
    else if (event == HOMEKIT_EVENT_CLIENT_CONNECTED) {
        if (!paired) //<- Here is the little unexpected comparison
            led_status_set(led_status, &pairing);
    }
    else if (event == HOMEKIT_EVENT_CLIENT_DISCONNECTED) {
        if (!paired)
            led_status_set(led_status, &unpaired);
    }
...
}
```
I would have expected at line 6:
```
  if (paired) // not (!paired)
      led_status_set(led_status, &pairing);
```