maximkulkin/esp-homekit #72: Led status example

GermanSheepDog, 2019-02-03:


GermanSheepDog, 2019-02-03:
I found a little challenge at led_status example (line 6):
```
void on_event(homekit_event_t event) {
    if (event == HOMEKIT_EVENT_SERVER_INITIALIZED) {
        led_status_set(led_status, paired ? &normal_mode : &unpaired);
    }
    else if (event == HOMEKIT_EVENT_CLIENT_CONNECTED) {
        if (!paired) //<- Here is the little unexpected comparison
            led_status_set(led_status, &pairing);
    }
    else if (event == HOMEKIT_EVENT_CLIENT_DISCONNECTED) {
        if (!paired)
            led_status_set(led_status, &unpaired);
    }
...
}
```
I would have expected at line 6:
```
  if (paired) // not (!paired)
      led_status_set(led_status, &pairing);
```

maximkulkin, 2019-02-11:
No, the code is correct: when client connects, you get HOMEKIT_EVENT_CLIENT_CONNECTED event. By that time it's just socket connection, but if there are no pairings yet, there is a big chance that client has connected to pair (and thus transitioning to pairing state). If client has connected and there is a pairing already, then do not do anything: by that time pairing is disabled and client will try pairing and fail or it will do pair verification and proceed managing accessory as usual.

closed 2019-02-11 by maximkulkin