maximkulkin/esp-homekit #77: Programmable Switch Event

GermanSheepDog, 2019-03-03:
Hello Maxim,

I worked with esp_homekit and the services
- HOMEKIT_SERVICE_SWITCH
- HOMEKIT_SERVICE_LIGHTBULB
- HOMEKIT_SERVICE_CONTACT_SENSOR  
- HOMEKIT_SERVICE_TEMPERATURE_SENSOR 
- HOMEKIT_SERVICE_HUMIDITY_SENSOR 
- HOMEKIT_SERVICE_MOTION_SENSOR
- HOMEKIT_SERVICE_CARBON_DIOXIDE_SENSOR
- HOMEKIT_SERVICE_AIR_QUALITY_SENSOR

and everything was fine.
Now I need to implement HOMEKIT_SERVICE_STATELESS_PROGRAMMABLE_SWITCH with HOMEKIT_CHARACTERISTIC_PROGRAMMABLE_SWITCH_EVENT.

I got an error: ">>> HomeKit: [Client 54] Remove Pairing"

The details of my analysis are:
```
>>> HomeKit: [Client 54] Verification successful, secure session established
>>> HomeKit: [Client 54] Get Accessories
START homekit_server_on_get_accessories
actor_getter: 10 0x25  0                        On Off                   bool 
actor_getter: 13 0x25  0                        On Off                   bool 
actor_getter: 16 0x25  0                        On Off                   bool 
actor_getter: 19 0x25  0                        On Off                   bool 
actor_getter: 22 0x25  1                        On Off                   bool 
actor_getter: 25 0x25  2                        On Off                   bool 
actor_getter: 57 0x73  0 Programmable Switch Event 0                     uint8 
ENDE homekit_server_on_get_accessories
homekit_server_on_pairings Status=4  //=TLVMethod_RemovePairing!
>>> HomeKit: [Client 54] Remove Pairing
>>> HomeKit: Removed pairing with A6836AB0-B4AF-4F15-9AB4-E3FE1FE823DF
```
After the function"actor_getter" for the " Programmable Switch Event" the error appeared. I think that the data type uint8 and the value of 0 is ok.
Where can I find some more information of the error? I like to see the answer from my Apple Device.
Can you help me?

GermanSheepDog, 2019-03-04:
Hello,
I found my first error in "Programmable Switch Event". It was a copy&paste error: The name of the service was not correct.
But in future it would help me, if I can see the info which was sent to my Apple Device and if I can see the answer.

maximkulkin, 2019-03-05:
@GermanSheepDog you can enable debug output with 
```shell
export HOMEKIT_DEBUG=1
```
before firmware compilation.

closed 2019-03-05 by maximkulkin