maximkulkin/esp-homekit #79: esp8266 add device failed. Write value too large [json.c]

LiangyongFShen, 2019-03-07:
>>> HomeKit: [Client 1] Verification successful, secure session established
>>> HomeKit: [Client 1] Get Accessories
!!! HomeKit: Write value too large
>>> HomeKit: [Client 1] Closing client connection

maximkulkin, 2019-03-07:
What are you trying to do?

LiangyongFShen, 2019-03-12:
Hello,  I  got some troubles!

- SDK: https://github.com/espressif/ESP8266_RTOS_SDK
- Board: esp8266

I did not change [json.c] anything. 

Someone say : 

change

```
void json_float(json_stream *json, float x) {
   ...
    void _do_write() {
        json_write(json, "%1.15g", x); 
    }
   ...
}
```
to 
```
void json_float(json_stream *json, float x) {
   ...
    void _do_write() {
        json_write(json, "%ld", x); 
    }
   ...
}
```

doesn't  work for me !!!

### Here my example code:

```
void json_float(json_stream *json, float x) {
   ...
    void _do_write() {
        json_write(json, "%ld", x); 
    }
   ...
}
```

```
#include <stdio.h>
#include <string.h>

#include <esp_system.h>
#include <esp_wifi.h>
#include <esp_wifi_types.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include <homekit/types.h>

#include "wf_homekit_header.h"

#define LED_ON 0                // this is the value to write to GPIO for led on (0 = GPIO low)
#define LED_INBUILT_GPIO 2      // this is the onboard LED used to show on/off only
#define LED_COUNT 16            // this is the number of WS2812B leds on the strip
#define LED_RGB_SCALE 255       // this is the scaling factor used for color conversion

// Global variables
float led_hue = 50;              // hue is scaled 0 to 360
float led_saturation = 50;      // saturation is scaled 0 to 100
int led_brightness = 50;     // brightness is scaled 0 to 100
uint32_t led_color_temp = 50;     // color temperate is scaled 50 to 100
bool led_on = false;            // on is boolean on or off
[... 227 lines cut ...]
D (47834) nvs: nvs_open_from_partition homekit 1
D (47834) nvs: nvs_get_str_or_blob _PAIRING_9_
D (47844) nvs: nvs_close 40
D (47844) nvs: nvs_open_from_partition homekit 1
D (47844) nvs: nvs_get_str_or_blob _PAIRING_10_
D (47854) nvs: nvs_close 41
D (47854) nvs: nvs_open_from_partition homekit 1
D (47854) nvs: nvs_get_str_or_blob _PAIRING_11_
D (47864) nvs: nvs_close 42
D (47864) nvs: nvs_open_from_partition homekit 1
D (47864) nvs: nvs_get_str_or_blob _PAIRING_12_
D (47874) nvs: nvs_close 43
D (47874) nvs: nvs_open_from_partition homekit 1
D (47884) nvs: nvs_get_str_or_blob _PAIRING_13_
D (47884) nvs: nvs_close 44
D (47884) nvs: nvs_open_from_partition homekit 1
D (47894) nvs: nvs_get_str_or_blob _PAIRING_14_
D (47894) nvs: nvs_close 45
D (47894) nvs: nvs_open_from_partition homekit 1
D (47904) nvs: nvs_get_str_or_blob _PAIRING_15_
D (47904) nvs: nvs_close 46
>>> HomeKit: Last admin pairing was removed, enabling pair setup
>>> HomeKit: Configuring mDNS
>>> HomeKit: [Client 1] Closing client connection
```

LiangyongFShen, 2019-03-12:
Hello again ! 

If you don't change the code 

```
void json_float(json_stream *json, float x) {
   ...
    void _do_write() {
        json_write(json, "%1.15g", x); 
    }
   ...
}
```
to
```
void json_float(json_stream *json, float x) {
   ...
    void _do_write() {
        json_write(json, "%ld", x); 
    }
   ...
}
```

and kept the  code as:

```
void json_float(json_stream *json, float x) {
   ...
    void _do_write() {
        json_write(json, "%1.15g", x); 
    }
   ...
}
```

and the  LED example code:

```
#include <stdio.h>
#include <string.h>

#include <esp_system.h>
#include <esp_wifi.h>
#include <esp_wifi_types.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include <homekit/types.h>

#include "wf_homekit_header.h"

#define LED_ON 0                // this is the value to write to GPIO for led on (0 = GPIO low)
#define LED_INBUILT_GPIO 2      // this is the onboard LED used to show on/off only
#define LED_COUNT 16            // this is the number of WS2812B leds on the strip
#define LED_RGB_SCALE 255       // this is the scaling factor used for color conversion

// Global variables
float led_hue = 50;              // hue is scaled 0 to 360
float led_saturation = 50;      // saturation is scaled 0 to 100
int led_brightness = 50;     // brightness is scaled 0 to 100
uint32_t led_color_temp = 50;     // color temperate is scaled 50 to 100
bool led_on = false;            // on is boolean on or off

void led_identify(homekit_value_t _value) {
    printf(">>> led_identify 1\n");
}

homekit_value_t led_on_get() {
    return HOMEKIT_BOOL(led_on);
[... 172 lines cut ...]
D (51665) nvs: nvs_open_from_partition homekit 1
D (51675) nvs: nvs_get_str_or_blob _PAIRING_0_
D (51675) nvs: nvs_close 25
D (51675) nvs: nvs_open_from_partition homekit 1
D (51675) nvs: nvs_set_blob _PAIRING_0_ 73
D (51685) nvs: nvs_close 26
>>> HomeKit: Added pairing with 092D5506-7CF1-45BB-8E8E-CDFF93758E51
>>> HomeKit: Configuring mDNS
>>> HomeKit: [Client 1] Successfully paired
>>> HomeKit: [Client 1] Closing client connection
>>> HomeKit: Got new client connection: 1
>>> HomeKit: [Client 1] Pair Verify Step 1/2
>>> HomeKit: [Client 1] Pair Verify Step 2/2
D (52465) nvs: nvs_open_from_partition homekit 1
D (52465) nvs: nvs_get_str_or_blob _PAIRING_0_
D (52465) nvs: nvs_close 27
>>> HomeKit: [Client 1] Found pairing with 092D5506-7CF1-45BB-8E8E-CDFF93758E51
>>> HomeKit: [Client 1] Verification successful, secure session established
.[0;32mI (55485) wf_example_main: Free heap size: 45720 for count: 5.[0m
>>> HomeKit: [Client 1] Get Accessories
!!! HomeKit: Write value too large
>>> HomeKit: [Client 1] Closing client connection
```

I don't know how to deal with it now ！！！

maximkulkin, 2019-03-12:
If you get lost, you can fallback to using ESP-OPEN-RTOS instead which is the official supported SDK for esp-homekit.

xxfenxx, 2019-03-13:
Thanks to @maximkulkin, but we have to use ESP8266_RTOS_SDK to do something，is it because they are different, you know what the reason might be ?

The error `!!! HomeKit: Write value too large` occur to is because [json.c] do not support float value in esp8266 ?  Can I replace `json_write(json, "%1.15g", x); ` with `json_write(json, "%ld", x); ` ?

closed 2019-03-13 by xxfenxx