maximkulkin/esp-homekit #81: !!! HomeKit: [Client 5] Failed to dump SPR public key (code -2)

emrahduruer, 2019-03-11:
Hello,

First time, I am trying to implement with this library. I am having pairing issues. Does anyone experience a similar or the same problem?

Please advice.

`>>> HomeKit: Configuring mDNS
>>> homekit_setup_mdns: Accessory Setup ID = UP00
mDNS announcement: Name=beryllium-FE38 ␓md=hermes-beryllium␆pv=1.0␔id=FE:38:D0:76:D9:66␄c#=1␄s#=1␄ff=0␄sf=1␄ci=8␋sh=rzy50Q== Port=5556 TTL=4500
>>> homekit_run_server: Staring HTTP server
>>> HomeKit: Got new client connection: 5
>>> homekit_client_process: [Client 5] Got 132 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 20792
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 5] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 20708
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 5] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 20044
>>> homekit_server_on_pair_setup: [Client 5] Using user-specified password: 006-28-942
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Failed to get SRP verifier (code -2)
>>> crypto_srp_get_public_key: Calculating public key
!!! HomeKit: [Client 5] Failed to dump SPR public key (code -2)
>>> send_tlv_response: [Client 5] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> tlv_debug: Type 7 value (1 bytes): \x01
>>> client_send: [Client 5] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 6\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x02\x07\x01\x01
>>> homekit_client_process: [Client 5] Finished processing
>>> HomeKit: [Client 5] Closing client connection
`

maximkulkin, 2019-03-11:
You’re running out of memory. I see that when pairing starts you have only ~20KB of RAM. Pairing process requires at least 23-24KB (that memory is used only during pairing and will be freed back afterwards. Code -2 during crypto actually means “out of memory”.

Normally with recent ESP8266 you have around 36-38KB of RAM so this is not a provlem, unless you allocate it for your own needs. I suggest postponing your accessory real initialization to when device has paired. You can use homekit_server_config_t.on_event callback and homekit_is_paired() to detect when pairing needs to be done and doing a “light” initialization of your accessory (the one that provides simplified behavior if any at all and does not require allocating a lot of memory).

emrahduruer, 2019-03-11:
Thank you. You solved my problem.

closed 2019-03-11 by maximkulkin

curla92, 2019-05-31:
hi maxim i have the same error. i have to postponed hardware init when homekit is paired? right? 
thanks

maximkulkin, 2019-06-01:
If you're getting error -2 during pairing process, that means you're running out of memory. You should postpone your other tasks startup to a point after pairing (after pairing you won't need that many memory to run HomeKit server) or use non-task-based solutions (timers/interrupt triggers). You can use homekit event callback to get notified when pairing is added/removed.

CHAITANYAPHADTARE02, 2025-01-15:
can someone please explain me how to do it