maximkulkin/esp-homekit #82: Multiple Accessory - Wifi

kunjal83, 2019-03-22:
Hi, 
I am facing compiling errors when I am adding multiple accessory. 
Is this how we do it? 

`HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]){
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Sample LED"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "037A2BABF19D"),
            HOMEKIT_CHARACTERISTIC(MODEL, "MyLED"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, led_identify),
            NULL
        }),        HOMEKIT_ACCESSORY(.id=2, .category=homekit_accessory_category_lightbulb, .service=(homekit_service_t*[]){
            HOMEKIT_SERICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
                HOMEKIT_CHARACTERISTIC(NAME, "Light Bulb 1"),
                HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
                HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "037A2BABF20D"),
                HOMEKIT_CHARACTERISTIC(MODEL, "Light Bulb 1"),
                HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
                HOMEKIT_CHARACTERISTIC(IDENTIFY, led_identify),
            }),`

Please help.

maximkulkin, 2019-03-22:
It would be nice if you have provided those compiler errors.. Also, your code snippet is incomplete.

kunjal83, 2019-03-22:
Apologies if it was not clear earlier. 

**Error:** 
/home/.../projects/esp-homekit-demo/examples/wifi_config/main.c:144:0: error: unterminated argument list invoking macro "HOMEKIT_ACCESSORY"
 }
 ^
/home/.../projects/esp-homekit-demo/examples/wifi_config/main.c:92:5: error: 'HOMEKIT_ACCESSORY' undeclared here (not in a function)
     HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]){
     ^

Complete Code for adding Accessories: 
`homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]){
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Sample LED"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "037A2BABF19D"),
            HOMEKIT_CHARACTERISTIC(MODEL, "MyLED"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, led_identify),
            NULL
        }),

     HOMEKIT_ACCESSORY(.id=2, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]){
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Light 1"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "037A2BABF20D"),
            HOMEKIT_CHARACTERISTIC(MODEL, "Light 1"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, led_identify),
            NULL
        }),

        HOMEKIT_SERVICE(LIGHTBULB, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Sample LED"),
            &led_on,
            NULL
        }),

        HOMEKIT_SERVICE(LIGHTBULB, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Light 1"),
            &led_on,
            NULL
            }),

        NULL
    }),
    NULL
};
`
I hope this helps. 

Thanks.

maximkulkin, 2019-03-22:
Right before `HOMEKIT_ACCESSORY(.id=2, ` you're missing another `NULL, }),` to close first accessory macro.

kunjal83, 2019-03-23:
Appreciate your help. This change did workout perfectly fine. 

I am not trying to figure out what does the 2 methods related to the identify "led_identify" and "led_identify_task" really function. if you happen to have any documentation around them, that would be really helpful.

maximkulkin, 2019-03-23:
HomeKit defines a way to identify an accessory after pairing, so you know which particular accessory you have paired. For that, it requires presence of write-only boolean characteristic (IDENTIFY) which should trigger some kind of activity so that user can, well, identify it.
So, since value of IDENTIFY characteristic does not matter, it's definition macro instead of initial value actually expects a setter function as a second argument. Since process of identification could be long (might for example requires blinking LED for a couple of seconds), it is better to do that asynchronously. That's why led_identify (which is a setter function and thus accepts a homekit_value_t, although nobody cares about the value, it's just the fact that value is being written) spawns a task (like a separate thread) that will handle identification procedure - led_identify_task. The task does a few iterations blinking an LED and then deletes itself.
Hope that helps

closed 2019-03-23 by maximkulkin