maximkulkin/esp-homekit #83: Fatal exception (29) caused when the second client is connected.

emrahduruer, 2019-03-22:
Hello,

I experience this problem;
As soon as the second client connected, I get cpu reset Fatal Exception(29) at 0x402410c2: uxListRemove at /tools/esp-open-rtos/FreeRTOS/Source/list.c:176.

So, I have read issue #13 and issue #10. I followed all the suggestions in there. No luck so far. I have the latest esp-homekit code including your March 14 update.

I tried to explain what I see so far. Please advise. 

1. In one scenerio, the second client connects and it works fine for a while.
`>>> HomeKit: Got new client connection: 5
[hermes-beryllium] []	on_homekit_event: HOMEKIT_EVENT_CLIENT_CONNECTED
>>> homekit_client_process: [Client 5] Got 165 incomming data
>>> homekit_server_on_pair_verify: HomeKit Pair Verify
>>> homekit_server_on_pair_verify: Free heap: 30504
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> tlv_debug: Type 3 value (32 bytes): s\xD4\xE8\xE4-\x9E\xED5xQ\x9Fxj\x1B\xD9\x8F7\x8A\xB8?\xC5\xBAq\xE8jHi\x93\x97\x8C\xBDB
>>> HomeKit: [Client 5] Pair Verify Step 1/2
>>> homekit_server_on_pair_verify: [Client 5] Importing device Curve25519 public key
>>> homekit_server_on_pair_verify: [Client 5] Generating accessory Curve25519 key
>>> homekit_server_on_pair_verify: [Client 5] Exporting accessory Curve25519 public key
>>> homekit_server_on_pair_verify: [Client 5] Generating Curve25519 shared secret
>>> homekit_server_on_pair_verify: [Client 5] Generating signature
>>> homekit_server_on_pair_verify: [Client 5] Generating proof
>>> homekit_server_on_pair_verify: [Client 5] Encrypting response
>>> send_tlv_response: [Client 5] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> tlv_debug: Type 3 value (32 bytes): *\xBA\x16UY\x93J\xC3\xB8/\x05\xE9\x01\xEF\x97\x0F\x09\xBFM\xF2S\xBE\x8D\xBB\x1E59\xE4\xCCW\x9Cg
>>> tlv_debug: Type 5 value (101 bytes): \x108\xD3\xE4\xDE\xB2\x1D20\x10\x1BA\xEA\xE4}\xE5DI\x0F\x0D\xF7\xFA\xEB2`\x9A'\x16\x80\xAD\x08\x14\x9E\xCD\xEF{\x915\xA1t#\xD4}EA\xAF\xFA\xF0\xFCb\xBCo\xE26\xA1\x8Ei\x10\x08\xC9\xB7~C\xBA\xFEY8\xB0\xB4\xF8\xE5^\x96\x128\x8F\xF1]\xD8X\x02\xBA-\xB2\xD6\xEFU3\x0F\x0C\x9DK)\xEEr\xD9\x9D \x8A\x0A\xFD
>>> client_send: [Client 5] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 140\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x02\x03 *\xBA\x16UY\x93J\xC3\xB8/\x05\xE9\x01\xEF\x97\x0F\x09\xBFM\xF2S\xBE\x8D\xBB\x1E59\xE4\xCCW\x9Cg\x05e\x108\xD3\xE4\xDE\xB2\x1D20\x10\x1BA\xEA\xE4}\xE5DI\x0F\x0D\xF7\xFA\xEB2`\x9A'\x16\x80\xAD\x08\x14\x9E\xCD\xEF{\x915\xA1t#\xD4}EA\xAF\xFA\xF0\xFCb\xBCo\xE26\xA1\x8Ei\x10\x08\xC9\xB7~C\xBA\xFEY8\xB0\xB4\xF8\xE5^\x96\x128\x8F\xF1]\xD8X\x02\xBA-\xB2\xD6\xEFU3\x0F\x0C\x9DK)\xEEr\xD9\x9D \x8A\x0A\xFD
>>> homekit_client_process: [Client 5] Finished processing
>>> homekit_client_process: [Client 5] Got 254 incomming data
>>> homekit_server_on_pair_verify: HomeKit Pair Verify
>>> homekit_server_on_pair_verify: Free heap: 30220
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 5 value (120 bytes): \x1B\xDD\xA5\xA3\xD4+\x09\x15j\xC0\x08\x87,Q1\x1B\xBF.\xCD\x9E\x8B\x8A\xA1\xC1b\x88\x00\xE4kiLx\xD9\x1B\x8B\xA4M\x18\xB0\xE4\x8C\x86\xBE$\xE8\x06\x95\xB8\xA3\xB8\xD5\xBF\xD7\x14\xB1\x8E\x04\x04\xB1k\xE1$\xAF\xA7\x044w\x1AfYVP\xBD\x0C'b\xD8\x80D@}\xEF\xAA\x8D/\xE03\x83\x8F\x90)3\x1F\x14\x8A/L\\\xBA\x10\xD3\xF5\xCC\x16\xF2\xD6\xE2'\xBC\x12\xDB\xDD\xE9\xE8\xDC:\xFB\xFD)7
>>> tlv_debug: Type 6 value (1 bytes): \x03
>>> HomeKit: [Client 5] Pair Verify Step 2/2
>>> homekit_server_on_pair_verify: [Client 5] Decrypting payload
>>> homekit_server_on_pair_verify: [Client 5] Searching pairing with F19D5199-BF7C-4BF7-B083-1C96550395FD
>>> HomeKit: [Client 5] Found pairing with F19D5199-BF7C-4BF7-B083-1C96550395FD
>>> homekit_server_on_pair_verify: [Client 5] Verifying device signature
>>> send_tlv_response: [Client 5] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x04
>>> client_send: [Client 5] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 3\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x04
[hermes-beryllium] []	on_homekit_event: HOMEKIT_EVENT_CLIENT_VERIFIED
>>> HomeKit: [Client 5] Verification successful, secure session established
>>> homekit_client_process: [Client 5] Finished processing
>>> homekit_client_process: [Client 5] Got 85 incomming data
>>> homekit_client_process: [Client 5] Decrypting data
>>> homekit_client_process: [Client 5] Decrypted 67 bytes, available 0
Decrypted data (67 bytes): "GET /accessories HTTP/1.1\x0D\x0AHost: beryllium-5C6A._hap._tcp.local\x0D\x0A\x0D\x0A"
>>> HomeKit: [Client 5] Get Accessories
>>> homekit_server_on_get_accessories: Free heap: 30400
>>> client_send: [Client 5] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/hap+json\x0D\x0ATransfer-Encoding: chunked\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A
>>> client_send: [Client 5] Sending payload: 3fe\x0D\x0A{"accessories":[{"aid":1,"services":[{"iid":1,"type":"3E","hidden":false,"primary":false,"characteristics":[{"aid":1,"iid":2,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"beryllium"},{"aid":1,"iid":3,"type":"20","perms":["pr"],"description":"Manufacturer","format":"string","value":"hermes-iot"},{"aid":1,"iid":4,"type":"30","perms":["pr"],"description":"Serial Number","format":"string","value":"037A2BABF19D"},{"aid":1,"iid":5,"type":"21","perms":["pr"],"description":"Model","format":"string","value":"hermes-beryllium"},{"aid":1,"iid":6,"type":"52","perms":["pr"],"description":"Firmware Revision","format":"string","value":"hermes-beryllium"},{"aid":1,"iid":7,"type":"14","perms":["pw"],"description":"Identify","format":"bool"}]},{"iid":8,"type":"49","hidden":false,"primary":true,"characteristics":[{"aid":1,"iid":9,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"Relay 1"},{"aid":1,"iid":10,"type":"25","perms":["pr","pw","ev"],"ev":false,"description":"On",\x0D\x0A
>>> client_send: [Client 5] Sending payload: 3fd\x0D\x0A"format":"bool","value":false}]},{"iid":11,"type":"49","hidden":false,"primary":false,"characteristics":[{"aid":1,"iid":12,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"Relay 2"},{"aid":1,"iid":13,"type":"25","perms":["pr","pw","ev"],"ev":false,"description":"On","format":"bool","value":true}]},{"iid":14,"type":"8A","hidden":false,"primary":false,"characteristics":[{"aid":1,"iid":15,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"Temperature Sensor"},{"aid":1,"iid":16,"type":"11","perms":["pr","ev"],"ev":false,"description":"Current Temperature","format":"float","unit":"celsius","minValue":0,"maxValue":100,"minStep":0.100000001490116,"value":23.8999996185303}]},{"iid":17,"type":"82","hidden":false,"primary":false,"characteristics":[{"aid":1,"iid":18,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"Humidity Sensor"},{"aid":1,"iid":19,"type":"10","perms":["pr","ev"],"ev":false,"description":"Current Relative Humidity","format":\x0D\x0A
>>> client_send: [Client 5] Sending payload: 196\x0D\x0A"float","unit":"percentage","minValue":0,"maxValue":100,"minStep":1,"value":31.7999992370605}]},{"iid":20,"type":"85","hidden":false,"primary":false,"characteristics":[{"aid":1,"iid":21,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"Motion Sensor"},{"aid":1,"iid":22,"type":"22","perms":["pr","ev"],"ev":false,"description":"Motion Detected","format":"bool","value":false}]}]}]}\x0D\x0A
>>> client_send: [Client 5] Sending payload: 0\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 5] Finished processing
[hermes-beryllium] []	homekit_gpi_callback: gpi_event_activated
>>> client_notify_characteristic: Got characteristic 1.22 change event
>>> client_notify_characteristic: Sending event to client 4
>>> send_client_events: [Client 4] Sending EVENT
>>> send_client_events: Free heap: 30452
>>> client_send: [Client 4] Sending payload: EVENT/1.0 200 OK\x0D\x0AContent-Type: application/hap+json\x0D\x0ATransfer-Encoding: chunked\x0D\x0A\x0D\x0A
>>> client_send: [Client 4] Sending payload: 35\x0D\x0A{"characteristics":[{"aid":1,"iid":22,"value":true}]}\x0D\x0A
[... 304 lines cut ...]
	"iid":	19,
	"ev":	true
}
>>> homekit_server_on_update_characteristics: [Client 4] Processing element {
	"aid":	1,
	"iid":	22,
	"ev":	true
}
[... cut ...] e
>>> client_send: [Client 4] Sending payload: HTTP/1.1 204 No Content\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 4] Finished processing
[hermes-beryllium] []	homekit__dht_data: homekit__dht_data: 23.0, 33.4
[hermes-beryllium] []	homekit__dht_data: Free heap: 26524
>>> client_notify_characteristic: Got characteri
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b
`

maximkulkin, 2019-03-22:
Can you show code of your motion detector?

emrahduruer, 2019-03-22:
Sure. Not that I was having the problem without the motion detector as well.

`homekit_value_t motion_get() {
    return HOMEKIT_BOOL(motion_detected);
}

homekit_characteristic_t motion = HOMEKIT_CHARACTERISTIC_(
    MOTION_DETECTED, false,
    .getter = motion_get);`
...
`    HOMEKIT_SERVICE(MOTION_SENSOR,
                    .characteristics = (homekit_characteristic_t*[]) {
        HOMEKIT_CHARACTERISTIC(NAME, "Motion Sensor"),
        &motion,
        NULL
    }),`
...
`void homekit_gpi_callback(uint8_t gpio_num, gpi_event event, void* cprm) {

        if (event == gpi_event_activated) {
            DEBUG_PRINT("gpi_event_activated");
            motion_detected = true;
            homekit_characteristic_notify(&motion, HOMEKIT_BOOL(motion_detected));
        } else if (event == gpi_event_deactivated) {
            DEBUG_PRINT("gpi_event_deactivated");
            motion_detected = false;
            homekit_characteristic_notify(&motion, HOMEKIT_BOOL(motion_detected));
        }
    }

    register_digital_sensor(MOTION_SENSOR_PIN, HIGH, &homekit_gpi_callback, NULL);`

maximkulkin, 2019-03-22:
Ok, I do not see anything criminal here, but last reset that you showed was caused by watchdog, so maybe you're doing too much in an interrupt or something. It would help if you'd show the whole code.

emrahduruer, 2019-03-22:
ok. here is the code where I control all my interrupts.
This code is calling the callback funtion in the homekit code to notify the clients.
DHT data changes is running in different task and calling another callback method in homekit code to notify.

`#include <common.h>

#include <stdio.h>
#include <FreeRTOS.h>
#include <task.h>
#include <esp/gpio.h>
#include <gpi.h>
#include <timers.h>

digital_gpi *gpis = NULL;

static void gpi_intr_callback_tick(digital_gpi *gpi) {
    gpi->prev_on_state_time = 0;
    gpi->callback(gpi->gpio_num,
                  gpi_event_single_press, gpi->callback_param);
}

/* The callback function that will execute in the context of the daemon task.
Note callback functions must all use this same prototype. */
void vProcessInterface( void *pvParameter1, uint32_t ulParameter2 ) {

    uint8_t gpio = *((uint8_t*) pvParameter1);
    /* ...Perform the processing here... */
    digital_gpi *gpi = find_gpi_by_gpio_num(gpio);
	if (gpi == NULL) {
		return;
    }

	uint32_t now = xTaskGetTickCountFromISR();
	if ((now - gpi->last_event_time) * portTICK_PERIOD_MS <
	    gpi->debounce_time) {
		// debounce time, ignore events
		return;
	}
	gpi->last_event_time = now;

    // Wait until the input changes...
    if (gpi->type == gpi_toggle_switch) {
        while(gpio_read(gpi->gpio_num) == gpi->last_state_value) {
        }
        gpi->last_state_value = gpio_read(gpi->gpio_num);
    }

	if (gpio_read(gpi->gpio_num) == gpi->on_state_value) {
        if (gpi->type == gpi_button) {
            gpi->prev_on_state_time = gpi->last_on_state_time;
        }
		gpi->last_on_state_time = now;

        if (gpi->type == gpi_digital_sensor) {
            // This is a sensor or any input...
            gpi->callback(gpi->gpio_num,
                          gpi_event_activated, gpi->callback_param);
        } else if (gpi->type == gpi_toggle_switch) {
            // This is a toggle switch...
            gpi->callback(gpi->gpio_num,
                          gpi_event_toggled_on, gpi->callback_param);
        }
	} else {
        if (gpi->type == gpi_button) {
            // This is a button...
            if (gpi->prev_on_state_time > 0 &&
                (gpi->last_on_state_time
                    - gpi->prev_on_state_time) * portTICK_PERIOD_MS <=
    		    DOUBLE_CLICK_TIME) {
                // Double click is detected.
[... 99 lines cut ...]

    return register_gpi(gpio_num,
                        gpi_toggle_switch,
                        on_state_value,
                        cb, cb_param);
}

digital_gpi* register_digital_sensor(uint8_t gpio_num,
                                     int on_state_value,
                                     gpi_callback cb, void* cb_param) {

    return register_gpi(gpio_num,
                        gpi_digital_sensor,
                        on_state_value,
                        cb, cb_param);
}

digital_gpi* find_gpi_by_gpio_num(uint8_t gpio_num) {
    digital_gpi *gpi = gpis;
    while (gpi && gpi->gpio_num != gpio_num) {
        gpi = gpi->next;
    }
    return gpi;
}
`

emrahduruer, 2019-03-22:
After you told me about the interrupt, while I am checking my code, I realized that in fact, the problem was the DHT data notifications. 

Thank you for your quick responses.

maximkulkin, 2019-03-22:
Consider it closed? Would you mind sharing your findings so that other can benefit?

closed 2019-04-01 by maximkulkin