maximkulkin/esp-homekit #87: PIL.Image

AchimPieters, 2019-04-26:
Hi,

I have a problem when trying to make a QR code for one of my devices. When I run your code as you describe it say's:

```
Traceback (most recent call last):
  File "tools/gen_qrcode", line 4, in <module>
    import PIL.Image
ImportError: No module named PIL.Image
```

How can I solve this? Thanks In advance!

AchimPieters, 2019-04-26:
okay I found out that I need to install pillow (pil) for this toi work:

However, since my project uses Python 3, No module named PIL error simply tells you that the python3 instance you're using cannot find the PIL library.

I used Homebrew for managing my system packages such as node, python, etc. So what I did was reinstall python3 with Homebrew with:

brew reinstall python, then run pip3 install pillow.

now I have Pillow-6.0.0 installed in the Phyton dir, but it still doesn't work?

maximkulkin, 2019-04-26:
I'm also running Python3 installed via Homebrew and it works fine. Make sure that default `python` executable resolves to that Homebrew Python3 (you might need to update your PATH) or just specify which python you want to use: `/usr/local/opt/python/libexec/bin/python tools/gen_qrcode ...`

AchimPieters, 2019-04-26:
Thank you for your quick reply:

I fixed half of my problem by doing the following:

`export PATH="/usr/local/opt/python/libexec/bin:$PATH"`

followed by:

`brew install python`

then I got:

```
Warning: python 3.7.3 is already installed, it's just not linked
You can use `brew link python` to link this version.
```
 and typed:

`brew link python`

Installed Pillow:

`pip install Pillow`

 then tried again:

`tools/gen_qrcode 1 123-45-678 1QJ8 qrcode.png`

now I'm getting:

```
Traceback (most recent call last):
  File "tools/gen_qrcode", line 7, in <module>
    import qrcode
ModuleNotFoundError: No module named 'qrcode'
```
so now I getting another error, If you have a pointer It would be welcome ;) Ill keep searching for the solution, keep you updated!

AchimPieters, 2019-04-26:
okay that was easy:

`pip install qrcode`

maximkulkin, 2019-04-26:
Hmm.. that's strange, I thought I added requirements.txt. Apparently I didn't.. I will fix that and update documentation.

AchimPieters, 2019-04-26:
that's great idea! :smile: see you around when I struggle with something else :wink:

closed 2019-04-26 by AchimPieters

TomKremer, 2020-01-22:
did run into the same issue, requirements.txt is still missing. Pillow & qrcode.

maximkulkin, 2020-01-22:
Fixed