maximkulkin/esp-homekit #88: esp8266 unreachable on idevices after aprox 2 min

michaelboke, 2019-05-09:
When i connect the iphone with the esp8266 running (example led or sonoff). It will pair and let me control the led i configured as output. This will work for a few minutes and after that the accessory becomes unreachable. IOS also reports that the accessory does not responds.

I have to hard reset the device and repair as a new accessory to have it responsive for again for the time it lasts. I tested it with 2 different esp8266 modules with 4mb flash.

I suspect that it has something todo with a timeout or when the connection disconnect i cannot connect anymore to the accessory.

Log of the first connect and pairing. The connection works and i can switch the led on the esp8266 on and off for testing. 
```
ESP-Open-SDK ver: 0.0.1 compiled @ May  9 2019 17:49:11
phy ver: 273, pp ver: 8.3

>>> wifi_config: Initializing WiFi config
>>> wifi_config: Found configuration, connecting to skynet
mode : sta(5c:cf:7f:1f:01:10)
add if0
scandone
add 0
aid 6
cnt

connected with skynet, channel 11
dhcp client start...
ip:192.168.1.162,mask:255.255.255.0,gw:192.168.1.1
>>> wifi_config: wifi_config_sta_connect_timeout_callback: Successfully connected
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 3B:7F:B9:6F:51:32
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Sonoff Outlet-1F0110-3B7F md=S20pv=1.0id=3B:7F:B9:6F:51:32c#=1s#=1ff=0sf=1ci=7 Port=5556 TTL=4500
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
>>> HomeKit: [Client 4] Pair Setup Step 3/3
>>> HomeKit: Added pairing with 64CC1B3B-8D56-4BFD-B0CC-82843B626F7A
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Sonoff Outlet-1F0110-3B7F md=S20pv=1.0id=3B:7F:B9:6F:51:32c#=1s#=1ff=0sf=0ci=7 Port=5556 TTL=4500
>>> HomeKit: [Client 4] Successfully paired
>>> HomeKit: Got new client connection: 5
>>> HomeKit: [Client 4] Closing client connection
>>> HomeKit: [Client 5] Pair Verify Step 1/2
>>> HomeKit: [Client 5] Pair Verify Step 2/2
>>> HomeKit: [Client 5] Found pairing with 64CC1B3B-8D56-4BFD-B0CC-82843B626F7A
>>> HomeKit: [Client 5] Verification successful, secure session established
>>> HomeKit: [Client 5] Get Accessories
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 5] Update Characteristics
Switch identify
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Closing client connection
```

After the `closing client connection` the esp8266 becomes unreachable, even when i soft reset the device (power cycle). The device is not reachable anymore.

The logs after the soft reset reset
```
ESP-Open-SDK ver: 0.0.1 compiled @ May  9 2019 17:49:11
phy ver: 273, pp ver: 8.3

>>> wifi_config: Initializing WiFi config
>>> wifi_config: Found configuration, connecting to skynet
mode : sta(5c:cf:7f:1f:01:10)
add if0
scandone
add 0
aid 6
cnt

connected with skynet, channel 11
dhcp client start...
ip:192.168.1.162,mask:255.255.255.0,gw:192.168.1.1
>>> wifi_config: wifi_config_sta_connect_timeout_callback: Successfully connected
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 3B:7F:B9:6F:51:32
>>> HomeKit: Found admin pairing with 64CC1B3B-8D56-4BFD-B0CC-82843B626F7A, disabling pair setup
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Sonoff Outlet-1F0110-3B7F md=S20pv=1.0id=3B:7F:B9:6F:51:32c#=1s#=1ff=0sf=0ci=7 Port=5556 TTL=4500
```

I am a bit lost here where to start searching. Who can help me getting to the source of the problem here.

michaelboke, 2019-05-09:
I narrowed it down to the point, when i pair (connect) the accessory and as long as i am connected i can communicate with the device. But as soon as i loose connection (just turning off wifi on the iphone) or the iphone goes to standby the accessory is not accessible anymore.

maximkulkin, 2019-07-10:
Does mDNS work correctly? Can you see accessory advertisement (`dns-sd -B _hap._tcp local` on macOS)?

nonameplum, 2019-08-02:
@maximkulkin I have a similar problem. I see that my esp8266 (Sonoff SV) is connected to the WiFi but is not visible with `dns-sd -B _hap._tcp`.
In my case is not always ~2 minutes. Sometimes is one day. Sometimes is a few minutes. This is quite random. Is there a way to investigate that?

maximkulkin, 2019-08-08:
@nonameplum what do logs show?

nonameplum, 2019-08-12:
@maximkulkin maybe it will be dumb question but is there a way to get logs while the device is connected to main power? Or I will need to implement some small storage based logging?

maximkulkin, 2019-08-12:
I personally debug connected to laptop, but I know people use [logging over UDP](https://github.com/HomeACcessoryKid/UDPlogger)

peros550, 2019-08-20:
I have some similar experience with only one device (Wemos D1 mini) running [this ](https://github.com/peros550/esp-homekit-multiple-sensors/tree/master/examples/multiple_sensors_ac), though another 2 devices running the same firmware on a different network & house seem to be stable for months. 

The device that seemed to be unresponsive nowadays was running error free for months using the same version of the above firmware. 

In my case, If I reboot the esp8266 device it gets back online. Furthermore, if I turn on & off the router's Wifi it also brings back online the device.

Because this error may appear after days of operation, it's very difficult to have logs. Unless of course these logs can be stored locally or sent remotely (i.e. to a raspberry). 

Thinking loudly, I was wondering, if at all is possible for the esp8266 device to be aware when itself becomes unavailable in the Home app?  If the answer is yes, it could be possible to implement  something to override the situation (force mDNS announcement or reboot) so that it can get back online to Home app.

nonameplum, 2019-08-20:
I was thinking to ping the router from the device from time to time to check if the connection is really alive. Not sure if that will verify the mDNS issue...

peros550, 2019-08-20:
> I was thinking to ping the router from the device from time to time to check if the connection is really alive. Not sure if that will verify the mDNS issue...

Be sure to check the latest version of esp-wifi-config which includes a monitoring capability of the network status. 

But I have observed a situation when the device was unreachable in Home app, but it was connected to the Wifi network.

nonameplum, 2019-08-20:
> But I have observed a situation when the device was unreachable in Home app, but it was connected to the Wifi network.

Exactly, I have that issue. I used wifi-config, and also implemented only that part of monitoring the network connection myself and it sounds like the wifi status is returning connected but the home app is losing connection to the device. I didn't have a chance yet to make some logging as I would need to have as @maximkulkin suggested UDP logging or storing the last couple of logs on the device.

d4rkmen, 2020-01-27:
Its weird, but HomeKit uses mDNS to check accessory online/offline status. Current responder implementation seems not stable.