maximkulkin/esp-homekit #93: HOMEKIT_SPI_FLASH_BASE_ADDR Question

kiralikbeyin, 2019-06-09:
@maximkulkin 
How do you decide HOMEKIT_SPI_FLASH_BASE_ADDR place?
My current config : HOMEKIT_SPI_FLASH_BASE_ADDR = 0x8c000 (im using https://github.com/RavenSystem/esp-homekit-devices)

I have a 660k firmware do you have any advice?

And
#define SPIFLASH_BASE_ADDR 0x200000 in homekit>src>storage.c do you think it is ok for a 660k firmware

kiralikbeyin, 2019-06-09:
I dont know logic of flash adresses but i found a solution for me.

0x8c000 equals to 573440
0x7A000 equals to 499712
0x8c000 - 0x7A000 difference is 73728

when i double the difference 
73728 * 2 = 147456 
147456 + 573440 = **720896** which means 0xb0000

0xb0000 goes to 720896 so that works. (my firmware 660000)

I think someone get help with this calculation.

closed 2019-06-09 by kiralikbeyin

maximkulkin, 2019-06-10:
The math there is: your flash size is 660000 which is 0xa1220, let's round it to 0xa2000. User firmware is written at address 0x2000 (you can see that in flash command: it flashes bootloader at 0x0, bootloader config at 0x1000 and user firmware goes at 0x2000). So overall your firmware will occupy address up to 0x2000 + 0xa2000 = 0xa4000. So you can use and address starting from 0xa4000 for storing settings. Of course it makes sense to leave some room for firmware to grow, so your 0xb0000 is perfectly fine. Keep in mind that Homekit settings use one full page of memory (4096 bytes or 0x1000).