maximkulkin/esp-homekit #94: esp-open-rtos

kiralikbeyin, 2019-06-11:
@maximkulkin 
Why do you use esp-open-rtos its last update was 6 months ago and there are a lot of issues.

Do you have any plan with https://github.com/espressif/ESP8266_RTOS_SDK

maximkulkin, 2019-06-11:
Maybe it not updated because it is already perfect. :)

By the time I started this project ESP_OPEN_RTOS was actively developed and I chose it over Espressif one because this one is OPEN SOURCE!

I do not have any plans to invest time into migration but would accept contributions if someone will do that.

kiralikbeyin, 2019-06-11:
I forgot about open source.. sorry..

I came here because no real life solution for websocket client or mqtt client in esp-open-rtos.

Can you show me how "esp as a client, httpS get with wolfssl" because http_get_mbedtls example eats heap like a monster and i cant go on..

maximkulkin, 2019-06-11:
I do not know what are your expectations about websockets and MQTT clients, but a quick search over EPS_OPEN_RTOS repository gave [this http server library](https://github.com/nochkin/libesphttpd/blob/master/include/libesphttpd/cgiwebsocket.h) which has websocket support and [this MQTT client library](https://github.com/SuperHouse/esp-open-rtos/tree/master/extras/paho_mqtt_c).

HTTPS client is a bit more tricky, I couldn't find an end-to-end example, but you can combine [this WolfSSL TLS example](https://github.com/wolfSSL/wolfssl-examples/blob/master/tls/client-tls-callback.c) with [HTTP parser library](https://github.com/SuperHouse/esp-open-rtos/tree/master/extras/http-parser)

kiralikbeyin, 2019-06-13:
@maximkulkin thanks a lot! 
I was fighting with HTTPS Client but my real problem was i forgot to close https connection...

TH10 was not responding after 3-4 turn on-off and i thought it is because of mbedtls function which eats heap. (i didnt use https calls no experience before)

Everything was fine when i remember to close https connection with old lib: https://github.com/maximkulkin/esp-homekit/commit/c2e67c468890b376343bf867309b31390af0dad9

Now i am using latest homekit lib but i cant pair which is really a heap problem:
````
RC > Saving last states
save_states()- heap  : 27840
>>> HomeKit: Generated new accessory key
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=RavenCore-88EA9C 
                                         md=RavenCorepv=1.0id=B6:82:6E:87:65:86c#=578s#=1ff=0sf=1ci=8
                                                                                                     sh=DMLFGQ== Port=5556 TTL=4500
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
!!! HomeKit: [Client 4] Failed to compute SRP shared secret (code -2)
>>> HomeKit: [Client 4] Closing client connection

`````
Latest heap i saw 27840

I tried these (make clean && make all ) in makefile
````
HOMEKIT_SPI_FLASH_BASE_ADDR = 0xb0000 # 720896
HOMEKIT_MAX_CLIENTS = 10 #also 8 12 and 16 not worked
HOMEKIT_SMALL = 1 # also 0
HOMEKIT_OVERCLOCK = 1 # also 0
HOMEKIT_OVERCLOCK_PAIR_SETUP = 1 # also 0
HOMEKIT_OVERCLOCK_PAIR_VERIFY = 1 # also 0
`````

other custom tasks are fired after 3 minutes later so they dont use extra heap when trying to pair.

Any advice?

maximkulkin, 2019-06-13:
27Kb should be enough. Try setting HOMEKIT_DEBUG=1 and see how much memory is there before each step.

kiralikbeyin, 2019-06-14:
@maximkulkin thanks for your support. I was trying ravencore which doesnt allow to change password and i changed it to more easier password. 
Normally i got !!! HomeKit: [Client 4] Failed to compute SRP shared secret (code -2)
Sorry my fault.

I didnt understand raven lib which needs EVE app to configure device type, it seems cool but difficult to manage all codes. I will go with https://github.com/maximkulkin/esp-homekit-demo

+It is good to know HOMEKIT_DEBUG=1  in makefile

My plan is to make virtual devices for homekit which talks with http enabled devices in the same network.

closed 2019-06-14 by kiralikbeyin