maximkulkin/esp-homekit #96: esp32 slow pairing

mentaluproar, 2019-06-29:
With the esp32 led example, I kept getting crashes.  After increasing the heap size for led_identify_task() to 1024, the crashes stopped but pairing now takes much longer.  Wolfssl4 makes use of the encryption acceleration features of the ESP32.  Is there a reason esp-homekit uses version 3.13?

maximkulkin, 2019-07-09:
There is no reason other than nobody have updated it to a new version. In my experience ESP32 pairing takes ~2s to pair and almost instant to do pair verify (establish a connection) which IMO is good enough.

LED identify task on ESP32 does need a bigger stack size now. I believe it is related to recent updates in ESP_IDF SDK as it worked before.

Give it another try. If you still experience slow pairing, please provide debug logs of pairing process and point to where it spends a lot of time.

mentaluproar, 2019-07-09:
I actually found a way around that. You need to explicitly set the processor speed. That fixes pairing.

closed 2019-07-09 by maximkulkin