maximkulkin/esp-homekit #98: Heap to small

GermanSheepDog, 2019-07-01:
Hello,
esp-homekit used a lot of the heap (for example: SERVER_TASK_STACK=12288) and sometimes it is a challenge.

I used a procedure which started every hour and which started a https-query. This query answered with a message with over 5000 bytes. So I got a memory problem with curl and lib/inflate.

My idea was: I like to stop homekit every hour and restart it after some seconds.

My approach:
1. Store status of homekit in server.c:
```
bool homekit_server_on = true;
bool is_homekit_server_stopped = false;
```
2. Stop server in proc homekit_run_server:
```
    //for (;;) {
    while (homekit_server_on) {
        fd_set read_fds;
        memcpy(&read_fds, &server->fds, sizeof(read_fds));
  
        struct timeval timeout = { 1, 0 }; /* 1 second timeout */
        int triggered_nfds = select(server->max_fd + 1, &read_fds, NULL, NULL, &timeout);
        if (triggered_nfds > 0) {
            if (FD_ISSET(server->listen_fd, &read_fds)) {
                homekit_server_accept_client(server);
                triggered_nfds--;
            }

            client_context_t *context = server->clients;
            while (context && triggered_nfds) {
                if (FD_ISSET(context->socket, &read_fds)) {
                    homekit_client_process(context);
                    triggered_nfds--;
                }

                context = context->next;
            }

            homekit_server_close_clients(server);
        }

        homekit_server_process_notifications(server);
    }

    server_free(server);
    is_homekit_server_stopped = true;
}
```
3. Create a stop procedure
```
bool homekit_stop_server_task() {
  homekit_server_on = false;
  uint8_t counter = 0;
  while ( (is_homekit_server_stopped == false) && (counter < 100) ) {
      counter++;
      vTaskDelay(100 / portTICK_RATE_MS);
  }
  return (counter < 100);
}
```
This worked fine. But how can I restart homekit without problems.

A seconds call of 'homekit_server_init' is not the correct solution.

maximkulkin, 2019-07-03:
Keep in mind that 12288 (bytes) is size for ESP32 which is rather small compared to how much memory ESP32 has (512KB, although almost half of that is taken by the system, it's still a decent amount of available memory). For ESP8266 it's only 2048 (words, which is 8192 bytes). Considering a basic ESP8266 example (e.g. an LED) has ~36K of RAM available not being able to generate a 5K response means that you have something else taking a lot of memory. I recommend exploring possibilities to reduce that part. E.g. HomeKit server itself tries to minimize amount of memory it uses when it generates it's responses. E.g. for JSON generation it has a custom manual generation which allows to generate JSON in order as it would appear in output and thus it can send chunked responses as they are being generated.

GermanSheepDog, 2019-07-04:
Hello Maxim,
of your point of view I absolutely agree.
But the primary problem is not the complete memory. It is "only" the heap with this limit: "Due to a technical limitation, the maximum statically allocated DRAM usage is 160KB. The remaining 160KB (for a total of 320KB of DRAM) can only be allocated at runtime as heap."
My heap-memory is divided in at least three parts:
1. The homekit part: It used very little part of heap.
2. My application part now (!) used a very little part of heap.
3. The ESP32-libraries used some times much memory.
One example for extensive disk heap usage is the curl-library.  If I got a request with a message length with 5800 bytes it used 5800 bytes of heap! It did not divide it into chunks! I looked at google for an alternative library for a HTTPS calls, but until now I did not found it.
My workaround today was to start at first time the HTTPS call and store the answer. I reboot and at second time a start homekit und my application part. I it not so nice but it worked.

maximkulkin, 2019-07-08:
Help me understand that: you're saying that you have more-or-less 160KB of RAM, out of which neither homekit nor your application does not use much. Then there is a, lets say, curl library that (given you have majority of those 160KB left available) can not fit 5800 bytes. Is that correct?

GermanSheepDog, 2019-07-09:
Hello Maxim,
I did not express myself exactly enough:
- I measured heap at my main loop and there are 160kb free.
- I did not know what curl did, but I know that my message its 5800 bytes long.
- I suspected that the all my esp-idf-libraries need a lot of memory.
- With 10k more heap it worked.

maximkulkin, 2019-07-09:
Try using a different HTTP client library. cURL is a generic library, I doubt it is intended to be used in embedded devices.

GermanSheepDog, 2019-07-09:
This was the first step I do. I found many HTTP client libraries but I need a HTTPS client library. I found everytime a solution with curl!

closed 2020-01-17 by maximkulkin