maximkulkin/esp-homekit #102: Accessory code

ericklil, 2019-08-19:
Thank you for your work!

If I wanted to change the `accessory` to say a Light with `HSV` values, Or make this control a fan with speed, direction etc, basically adding `characteristics` to the `accessory`. How would one find the setup code for that.

maximkulkin, 2019-08-19:
Well, to implement an accessory you need to add apropriate *services* which in turn have proper *characteristics*. Setup codes are either hard-coded inside firmware (and printed somewhere on packaging or on the accessory itself),
[image]

[image]

or you add a screen to accessory and show randomly generate setup code or even QR code.

[image]

[image]

ericklil, 2019-08-20:
Thank you for your quick reply, I erred in my question
I meant to ask; how to create an accessory other than the sample you provided. i.e a Light with HSV values such as a Neopixel or a Fan with speed, direction and other `characteristics or services`?
Where to find the `accessory category code` as you mention in your`readme.md`
In the sample is the `.setupId="1QJ8"`  random generated?

maximkulkin, 2019-08-20:
Well, the main read is [Homekit Accessory Protocol Specification](https://developer.apple.com/homekit/specification/) (you need to create an Apple Developer account to access it). You can find it on this page: https://developer.apple.com/homekit/

In esp-homekit all services and characteristics for the spec are defined in [homekit/characteristics.h](https://github.com/maximkulkin/esp-homekit/blob/master/include/homekit/characteristics.h). For convenience, service definitions in that header file are provided with list of required and optional characteristics.

closed 2019-09-26 by maximkulkin