maximkulkin/esp-homekit #109: Try to build ESP32 LED example on windows ... problem

NHBavaria, 2019-10-11:
First of all, I can successfull create the ESP-IDF examples. So I think I've set up the environment correctly.  Now I get an error when trying to create the esp32 LED demo of esp-homekit. Does anyone have an idea?

```
esp-homekit-demo\examples\esp32\led>idf.py build
....
[1/5] Performing build step for 'bootloader'
ninja: no work to do.
[2/3] Linking CXX executable redlabs_bridge.elf
FAILED: redlabs_bridge.elf
cmd.exe /C "cd . && C:\Users\Nico\.espressif\tools\xtensa-esp32-elf\1.22.0-80-g6c4433a5-5.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls  -nostdlib @CMakeFiles\redlabs_bridge.elf.rsp  -o redlabs_bridge.elf  && cd ."
esp-idf/homekit/libhomekit.a(base64.c.obj): In function `base64_encode':
C:/Users/Nico/Desktop/esp-idf/components/homekit/src/base64.c:39: multiple definition of `base64_encode'
esp-idf/wpa_supplicant/libwpa_supplicant.a(base64.c.obj):C:/Users/Nico/Desktop/esp-idf/components/wpa_supplicant/src/wpa2/utils/base64.c:31: first defined here
esp-idf/homekit/libhomekit.a(base64.c.obj): In function `base64_decode':
base64.c:(.text.base64_decode+0x0): multiple definition of `base64_decode'
esp-idf/wpa_supplicant/libwpa_supplicant.a(base64.c.obj):base64.c:(.text.base64_decode+0x0): first defined here
esp-idf/wolfssl/libwolfssl.a(sha256.c.obj):(.literal.InitSha256+0x0): undefined reference to `esp_sha_hw_unlock'
esp-idf/wolfssl/libwolfssl.a(sha256.c.obj):(.literal.wc_Sha256Update+0x4): undefined reference to `esp_sha_try_hw_lock'
esp-idf/wolfssl/libwolfssl.a(sha256.c.obj):(.literal.wc_Sha256Update+0x8): undefined reference to `esp_sha256_process'
esp-idf/wolfssl/libwolfssl.a(sha256.c.obj):(.literal.wc_Sha256Final+0x0): undefined reference to `esp_sha256_digest_process'
esp-idf/wolfssl/libwolfssl.a(sha256.c.obj): In function `InitSha256':
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha256.c:1065: undefined reference to `esp_sha_hw_unlock'
esp-idf/wolfssl/libwolfssl.a(sha256.c.obj): In function `Sha256Update':
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha256.c:1065: undefined reference to `esp_sha_try_hw_lock'
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha256.c:1065: undefined reference to `esp_sha256_process'
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha256.c:1065: undefined reference to `esp_sha_try_hw_lock'
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha256.c:1065: undefined reference to `esp_sha256_process'
esp-idf/wolfssl/libwolfssl.a(sha256.c.obj): In function `Sha256Final':
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha256.c:981: undefined reference to `esp_sha_try_hw_lock'
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha256.c:986: undefined reference to `esp_sha256_process'
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha256.c:1035: undefined reference to `esp_sha_try_hw_lock'
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha256.c:1040: undefined reference to `esp_sha256_digest_process'
esp-idf/wolfssl/libwolfssl.a(sha512.c.obj):(.literal.wc_Sha512Update+0x0): undefined reference to `esp_sha512_process'
esp-idf/wolfssl/libwolfssl.a(sha512.c.obj):(.literal.wc_Sha512Final+0x0): undefined reference to `esp_sha512_digest_process'
esp-idf/wolfssl/libwolfssl.a(sha512.c.obj): In function `InitSha512':
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha512.c:842: undefined reference to `esp_sha_hw_unlock'
esp-idf/wolfssl/libwolfssl.a(sha512.c.obj): In function `Sha512Update':
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha512.c:842: undefined reference to `esp_sha_try_hw_lock'
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha512.c:842: undefined reference to `esp_sha512_process'
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha512.c:842: undefined reference to `esp_sha_try_hw_lock'
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha512.c:842: undefined reference to `esp_sha512_process'
esp-idf/wolfssl/libwolfssl.a(sha512.c.obj): In function `Sha512Final':
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha512.c:756: undefined reference to `esp_sha_try_hw_lock'
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha512.c:758: undefined reference to `esp_sha512_process'
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha512.c:804: undefined reference to `esp_sha_try_hw_lock'
C:/Users/Nico/Desktop/esp-idf/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/sha512.c:806: undefined reference to `esp_sha512_digest_process'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

maximkulkin, 2019-10-11:
I do not use and/or care to support CMake at the moment. Try building it with regular make: `make all`

NHBavaria, 2019-10-11:
> I do not use and/or care to support CMake at the moment. Try building it with regular make: `make all`

Thank you! It works now.

closed 2019-10-11 by NHBavaria