maximkulkin/esp-homekit #110: Sort order of multiple HOMEKIT_SERVICE under one device card due to iOS13.

jihao, 2019-10-22:
Is it possible to sort multiple HOMEKIT_SERVICE appearing order? Because after I upgrade to iOS13, it becomes one long list card. 

My accessory define looks like this, but it's not shown as the defined order.  Maybe its an iOS limitation. I am just asking if it's possible to do the sort magic with esp-homekit. 

```
homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id = 1, .category = homekit_accessory_category_lightbulb, .services = (homekit_service_t*[]) {
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics = (homekit_characteristic_t*[]) {
            &name,
            &manufactor,
            &serial_number,
...
        }),
        HOMEKIT_SERVICE(LIGHTBULB, .primary = true, .characteristics = (homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "My LED"),
            &lightbulb_on,
            &lightbulb_brightness,
...
            NULL
        }),
        HOMEKIT_SERVICE(LIGHTBULB, .primary = false, .characteristics = (homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Effect Speed"),
            &lightbulb_fx_on,
           ...
            NULL
        }),
        HOMEKIT_SERVICE(LIGHTBULB, .primary = false, .characteristics = (homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Fireworks"),
           ...
            NULL
        }),
        HOMEKIT_SERVICE(LIGHTBULB, .primary = false, .characteristics = (homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Christmas"),
           ...
            NULL
        }),
        HOMEKIT_SERVICE(LIGHTBULB, .primary = false, .characteristics = (homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Halloween"),
           ...
            NULL
        }),
```

maximkulkin, 2019-10-22:
Could you describe what do you expect vs what you actually get. A screenshot maybe?

jihao, 2019-10-23:
Thanks for answering.  I added an image in the question.

**What I expect:**  fixed order like 1. "Fireworks", 2. "Christmas", 3. "Halloween"
**What I get:** I figured out that homekit is showing the names in alphabeta order. 

Any chance I can get fixed order in code? If not, that's also OK. I can live with alphabeta order.

maximkulkin, 2019-10-23:
Well, looks like the only way you can enforce sorting is to put an index right into name

timoschilling, 2019-10-24:
With iOS 13.2 you will get the ability to split this into single tiles like it was pre iOS 13

timoschilling, 2019-10-24:
To fix it right now it maybe work if you change the category of the accessory to an bright. The each characteristic should have his own tile.

jihao, 2019-10-25:
Thank you guys. I will close the issue.

closed 2019-10-25 by jihao