maximkulkin/esp-homekit #112: mDNS Sends Remove Command after ~2 Minutes - esp32 - No Response

AidanLovelace, 2019-11-07:
I can turn the device (a lamp accessory on an esp32) on and off for anywhere from 20s to 5min after the esp32 restarts, then iOS says No Response, so I checked the mDNS advertisement as I've seen in other issues and got the following:
```
$ dns-sd -B _hap._tcp local
Browsing for _hap._tcp.local
DATE: ---Wed 06 Nov 2019---
21:28:50.509  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
21:28:50.509  Add        2   7 local.               _hap._tcp.           Tall Lamp
21:28:59.058  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
```
It appears to me that it's advertising itself on the network, then retracting the advertisement for some reason. It's possible there's something wrong with my C code, so I'll attach that. I am on the latest master commit of [esp-homekit](https://github.com/maximkulkin/esp-homekit/commit/0c5139f192adf24081a3e11a96d440e93b14b451). And this commit of the [esp-idf](https://github.com/espressif/esp-idf/commit/758db1e008d4a524b4f386367760b6259e6d6737). 

Chip Information from the flash log:
```
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
```

My C code:
```c
#include <stdio.h>
// Was testing using wifi manager, but stopped for debugging.
//#include <wifi_manager.h>
#include <esp_wifi.h>
#include <esp_event_loop.h>
#include <esp_log.h>
#include <nvs_flash.h>

#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <driver/gpio.h>

#include <driver/gpio.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include "wifi.h"

void on_wifi_ready();

esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
        case SYSTEM_EVENT_STA_START:
            printf("STA start\n");
            esp_wifi_connect();
            break;
        case SYSTEM_EVENT_STA_GOT_IP:
            printf("WiFI ready\n");
            on_wifi_ready();
            break;
        case SYSTEM_EVENT_STA_DISCONNECTED:
            printf("STA disconnected\n");
            esp_wifi_connect();
            break;
        default:
            break;
    }
    return ESP_OK;
}

static void wifi_init() {
    // Was testing using wifi manager, but stopped for debugging.

    // /* start the wifi manager */
	// wifi_manager_start();

	// /* register a callback as an example to how you can integrate your code with the wifi manager */
	// wifi_manager_set_callback(EVENT_STA_GOT_IP, &on_wifi_ready);
[... 483 lines cut ...]
>>> client_send: [Client 56] Sending payload: 35\x0D\x0A{"characteristics":[{"aid":1,"iid":10,"value":true}]}\x0D\x0A
>>> client_send: [Client 56] Sending payload: 0\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 58] Got 193 incomming data
>>> homekit_client_process: [Client 58] Decrypting data
>>> homekit_client_process: [Client 58] Decrypted 175 bytes, available 0
Decrypted data (175 bytes): "PUT /characteristics HTTP/1.1\x0D\x0AHost: Tall\\032Lamp._hap._tcp.local\x0D\x0AContent-Length: 50\x0D\x0AContent-Type: application/hap+json\x0D\x0A\x0D\x0A{"characteristics":[{"aid":1,"iid":10,"value":0}]}"
>>> HomeKit: [Client 58] Update Characteristics
>>> homekit_server_on_update_characteristics: Free heap: 211080
>>> homekit_server_on_update_characteristics: [Client 58] Processing element {
	"aid":	1,
	"iid":	10,
	"value":	0
}
>>> process_characteristics_update: [Client 58] Updating characteristic 1.10 with boolean false
>>> client_notify_characteristic: Got characteristic 1.10 change event
>>> client_notify_characteristic: Sending event to client 56
>>> homekit_server_on_update_characteristics: [Client 58] There were no processing errors, sending No Content response
>>> client_send: [Client 58] Sending payload: HTTP/1.1 204 No Content\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 58] Finished processing
>>> send_client_events: [Client 56] Sending EVENT
>>> send_client_events: Free heap: 209484
>>> client_send: [Client 56] Sending payload: EVENT/1.0 200 OK\x0D\x0AContent-Type: application/hap+json\x0D\x0ATransfer-Encoding: chunked\x0D\x0A\x0D\x0A
>>> client_send: [Client 56] Sending payload: 36\x0D\x0A{"characteristics":[{"aid":1,"iid":10,"value":false}]}\x0D\x0A
>>> client_send: [Client 56] Sending payload: 0\x0D\x0A\x0D\x0A
```

AidanLovelace, 2019-11-07:
Update: I left the mDNS command running all day today. It looks like it's re-adding itself every hour or so.
```
DATE: ---Thu 07 Nov 2019---
 0:25:12.982  Add        2   7 local.               _hap._tcp.           Tall Lamp
 0:28:16.042  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 1:26:09.457  Add        2   7 local.               _hap._tcp.           Tall Lamp
 1:34:56.323  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 1:37:42.915  Add        2   7 local.               _hap._tcp.           Tall Lamp
 1:42:49.296  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 1:45:05.517  Add        2   7 local.               _hap._tcp.           Tall Lamp
 1:48:42.371  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 1:52:33.722  Add        2   7 local.               _hap._tcp.           Tall Lamp
 1:53:32.685  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 1:53:59.406  Add        2   7 local.               _hap._tcp.           Tall Lamp
 1:54:59.192  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 1:55:38.119  Add        2   7 local.               _hap._tcp.           Tall Lamp
 1:59:42.329  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 2:00:48.846  Add        2   7 local.               _hap._tcp.           Tall Lamp
 2:01:32.518  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 2:03:54.759  Add        2   7 local.               _hap._tcp.           Tall Lamp
 2:04:50.777  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 2:05:59.774  Add        2   7 local.               _hap._tcp.           Tall Lamp
 2:06:55.382  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 2:10:48.566  Add        2   7 local.               _hap._tcp.           Tall Lamp
 2:11:50.554  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 2:23:47.518  Add        2   7 local.               _hap._tcp.           Tall Lamp
 2:24:44.844  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 2:43:12.018  Add        2   7 local.               _hap._tcp.           Tall Lamp
 2:43:35.565  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 3:26:09.568  Add        2   7 local.               _hap._tcp.           Tall Lamp
 3:26:42.231  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 4:17:51.809  Add        2   7 local.               _hap._tcp.           Tall Lamp
 4:23:13.689  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 4:23:36.489  Add        2   7 local.               _hap._tcp.           Tall Lamp
 4:25:24.670  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 4:27:16.035  Add        2   7 local.               _hap._tcp.           Tall Lamp
 4:30:42.798  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 4:32:50.399  Add        2   7 local.               _hap._tcp.           Tall Lamp
 4:33:18.123  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 4:40:36.267  Add        2   7 local.               _hap._tcp.           Tall Lamp
 4:40:53.496  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 4:48:10.241  Add        2   7 local.               _hap._tcp.           Tall Lamp
 4:49:05.369  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 4:53:51.841  Add        2   7 local.               _hap._tcp.           Tall Lamp
 4:56:09.407  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 5:18:03.673  Add        2   7 local.               _hap._tcp.           Tall Lamp
 5:19:02.580  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 5:20:35.156  Add        2   7 local.               _hap._tcp.           Tall Lamp
 5:21:32.646  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 5:28:46.543  Add        2   7 local.               _hap._tcp.           Tall Lamp
 5:31:01.648  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 5:48:56.225  Add        2   7 local.               _hap._tcp.           Tall Lamp
 5:49:43.536  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 6:15:54.338  Add        2   7 local.               _hap._tcp.           Tall Lamp
 6:19:04.200  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 7:00:25.621  Add        2   7 local.               _hap._tcp.           Tall Lamp
 7:24:45.193  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 7:52:33.234  Rmv        0   7 local.               _hap._tcp.           Rachio-2961D6
 7:53:01.979  Add        2   7 local.               _hap._tcp.           Rachio-2961D6
 8:20:37.137  Add        2   7 local.               _hap._tcp.           Tall Lamp
 8:21:40.583  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 8:23:43.176  Add        2   7 local.               _hap._tcp.           Tall Lamp
 8:24:20.468  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 8:57:13.677  Add        2   7 local.               _hap._tcp.           Tall Lamp
 8:57:49.315  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
 9:32:18.014  Add        2   7 local.               _hap._tcp.           Tall Lamp
 9:32:58.026  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
[... cut ...]
16:22:55.755  Add        2   7 local.               _hap._tcp.           Tall Lamp
16:24:24.841  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
16:40:54.986  Add        2   7 local.               _hap._tcp.           Tall Lamp
16:41:53.534  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
16:46:12.085  Add        2   7 local.               _hap._tcp.           Tall Lamp
16:46:33.140  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
```

maximkulkin, 2019-11-08:
Can you do packet capture with Wireshark and make sure it actually sends MDNS packet or it just somehow expires. Usually mDNS is set up with 4500 seconds to expire and with same period it will reannounce itself.

Can you uncomment [this line](https://github.com/maximkulkin/esp-homekit/blob/master/src/mdnsresponder.c#L44) and see more log on what is happening inside mDNS.

AidanLovelace, 2019-11-09:
Okay, so I now have it running on two different devices. I did a pcap w/ WireShark and found that neither are sending packets. However, both drop off at the exact same time.

```
21:45:58.301  Add        2   7 local.               _hap._tcp.           Smart Plug
22:00:11.631  Add        2   7 local.               _hap._tcp.           Tall Lamp
22:04:35.195  Rmv        1   7 local.               _hap._tcp.           Smart Plug
22:04:35.195  Rmv        0   7 local.               _hap._tcp.           Tall Lamp
```

After uncommenting that line, I received no additional output from mdns. :/

fuatsengul, 2019-12-18:
@AidanLovelace did you find solution?

d4rkmen, 2020-01-17:
It seems the mDNS remove issue faced not only me! Probably the library issue

AidanLovelace, 2020-01-17:
I ended up switching to an ESP-8266 which does not have this issue.

nonameplum, 2020-01-21:
@AidanLovelace @maximkulkin I have very similar issue on ESP8266MOD 4mb (if that matters).
In my case I observe that frequently after `Rmv` devices do not announce themselves again in short period of time (quite random and hard to measure). But after some time I see again `Add` events and then process shall be repeated.
Example:
```
14:23:58.728  Add        3   4 local.               _hap._tcp.           Garage Inside
14:23:58.728  Add        2   4 local.               _hap._tcp.           Garage Door
14:24:12.884  Rmv        1   4 local.               _hap._tcp.           Garage Inside
14:24:12.884  Rmv        0   4 local.               _hap._tcp.           Garage Door
14:24:15.047  Add        2   4 local.               _hap._tcp.           Garage Door
14:24:17.046  Add        2   4 local.               _hap._tcp.           Garage Inside
14:47:25.499  Rmv        0   4 local.               _hap._tcp.           Garage Inside
15:02:46.522  Add        2   4 local.               _hap._tcp.           Garage Inside
```

If we focus only the device `Garage Inside` (added first column with the time diff to previous event):
```
		    14:23:58.728  Add        3   4 local.               _hap._tcp.           Garage Inside
	14s		14:24:12.884  Rmv        1   4 local.               _hap._tcp.           Garage Inside
	4s		14:24:17.046  Add        2   4 local.               _hap._tcp.           Garage Inside
	23m 8s	14:47:25.499  Rmv        0   4 local.               _hap._tcp.           Garage Inside
	15m 21s	15:02:46.522  Add        2   4 local.               _hap._tcp.           Garage Inside
```

We can see that time interval between the last two events `Rmv` and `Add` is 15 minutes and 12 sec. For that long devices in my case was not accessible by the home app and at the same time I could ping the device both by IP and by mDNS name `ping Garage\ Inside.local`.

In addition. I have macbook paired with my iPhone setup according to the home configuration. What I also observed is that sometimes one of the devices is available on macbook but not on the iPhone. After some time both devices might be not available on macbook and iPhone, then again both available on iPhone and macbook. I even encounter a case when I had opened the home app on macbook and iPhone, both devices were accessible from the macbook, but only one on iPhone. After killing the home app on the macbook previously unavailable device on iPhone became instantly available.

TomKremer, 2020-01-23:
I'm having the same trouble using my shelly 1 v3 (esp8266ex). Adding and removing all the time. Was using RavenSystem HAA, then compiled and flashed esp-homekit alone, same issue as on HAA. Very strange.

nonameplum, 2020-01-23:
I'm not familiar that much with mDNS machinery.
Seems like somehow the issue is only related to `_hap` as mDNS for `.local` works fine. I will try to collect logs on the weekend from mDNS after unlocking debug logs: 
```c
// #define qDebugLog             // Log activity generally
// #define qLogIncoming          // Log all arriving multicast packets
// #define qLogAllTraffic        // Log and decode all mDNS packets
```
and cross-reference with `dns-sd -B _hap._tcp local` output.

d4rkmen, 2020-01-23:
It depends on network load (by mdns?) It may work ok for wifi router network, but fail for real life home network. I was tried 3 mdns implementations so far: lwip native, mdnsresponer ported to RTOS SDK and mdns SDK component. Always the same :( service discovery interrupts somehow (even wothout pairing to start)

TomKremer, 2020-01-24:
In my case it was due to wifi ap. Stupid Apple Express Wifi looks having trouble with mdns. I now tried a fritzbox & linksys without any issues.

d4rkmen, 2020-01-24:
IMO this could be lwip and/or mdns implementation issue. In my network open and rtos-sdk gives same result

nonameplum, 2020-01-24:
I tried on TP-Link and Linksys and the same issue.

d4rkmen, 2020-01-25:
Found mdns implementation that keeps pairing process ok. But now in logs there is paired successfully, but home app still pairing (sends pair setup step 1 again)
I guess I hacked something... Anyway, reason was mdns.

nonameplum, 2020-01-27:
This might be good learning about mDNS with proper HAP support https://github.com/esp8266/Arduino/pull/5442

d4rkmen, 2020-01-27:
@nonameplum can u test the PR #123 ?

nonameplum, 2020-01-27:
Sure, but I'm away from home and I don't have access to devices. I will check later this week.

nonameplum, 2020-02-11:
@d4rkmen I checked on three devices the PR and there's not much difference in my case. Still I see `RMV` events sometimes not followed by `ADD`. In the home app accessories are also not available at that time.

d4rkmen, 2020-02-11:
@nonameplum sorry to hear that. it seems esp32 build has other flows

nonameplum, 2020-02-11:
@d4rkmen All three on ESP8266.

d4rkmen, 2020-02-11:
> Still I see `RMV` events sometimes not followed by `ADD`

Ah, you mean WiFi network connect / disconnect?

nonameplum, 2020-02-11:
I mean records from `dns-sd -B _hap._tcp local`.

nonameplum, 2020-02-17:
@d4rkmen I exchanged the `mdnsresponder.h` usage in hometkit's `port.c` with `lwip/apps/mdns.h` that I copied to my esp-open-rtos from https://github.com/ourairquality/lwip/tree/esp-open-rtos/src/apps/mdns . Looks optimistic for now but I'm still testing it.

nonameplum, 2020-02-19:
@d4rkmen I made a fork with exchanged `mdnsresponder.h` to use mentioned above lwip's mdns implementation. Could you check if it will work for you? For me its seems like availability of devices is more stable. I made also a test change in `server.c` to no set the flag `client->disconnect` as I encounter sometimes an issue that some connections get into the loop of `connect -> process -> disconnect` instead of staying connected which ends up with unavailable device in the home app even though mDNS works as expected, and the device is visible by `dns-sd -B _hap._tcp local`.