maximkulkin/esp-homekit #115: Not Responding on ESP8266

SmartArduino, 2019-12-11:
Running the homekit light demo on esp8266 about 2 hours.
The Home app shows the  "No response" against the lights.
My phone's IP address is 192.168.9.187.
The log show
mDNS IPv4 got 744 bytes from 192.168.9.187
Query, ID $0 op 0: 6 questions, 6 answers, 1 extra RR
 Q1: Switch-D812A7-95BF._hap._tcp.local. TXT  In
 Q2: Switch-ADDFB8-2C9F._hap._tcp.local. TXT  In
 Q3: Switch-89172F-D584._hap._tcp.local. TXT  In
 Q4: Light-D17D8F-98EF._hap._tcp.local. TXT  In
 Q5: Light-9A1CDB-60CC._hap._tcp.local. TXT  In
 Q6: Switch-8330FA-3DE7._hap._tcp.local. TXT  In
 A1: Switch-D812A7-95BF._hap._tcp.local. TXT  In TTL 4474 md=HomeKit-PA
 A2: Switch-ADDFB8-2C9F._hap._tcp.local. TXT  In TTL 4474 md=HomeKit-PA
 A3: Switch-89172F-D584._hap._tcp.local. TXT  In TTL 4474 md=HomeKit-PA
 A4: Light-D17D8F-98EF._hap._tcp.local. TXT  In TTL 4487 md=HomeKit-BA
 A5: Light-9A1CDB-60CC._hap._tcp.local. TXT  In TTL 4474 md=HomeKit-BA
 A6: Switch-8330FA-3DE7._hap._tcp.local. TXT  In TTL 4474 md=HomeKit-PC
 ExRR1:  type 41 class 1440 TTL 4500 18b:00 04 00 0E 00 0C A2 ED 2C 8A DD B3 80 ED 2C 8A DD B3 
 - matched 'Light-9A1CDB-60CC._hap._tcp.local.' TXT 
Response, ID $0 Auth  noerr: 0 questions, 1 answers
 A1: Light-9A1CDB-60CC._hap._tcp.local. TXT  In TTL 4500 md=HomeKit-BA
 - responded with 135 bytes err 0

 The Light-9A1CDB-60CC is my device.
The log display device received the mDNS request from the mobile phone and answered it correctly,
And I tested port 5556, which can also provide HTTP service normally.
However still no response in Home App.
I restart the devices. That's ok.

d4rkmen, 2020-01-27:
@SmartArduino in order to trace the issue u need to enable homekit debug output and log it for the fail moment

nonameplum, 2020-02-19:
You can try with my fork `https://github.com/nonameplum/esp-homekit/tree/my_master`. I changed the mDNS implementation to use mDNS from https://github.com/ourairquality/lwip/tree/esp-open-rtos/src/apps/mdns . I'm still testing it but seems like availability of the devices is more stable.

d4rkmen, 2020-02-19:
@nonameplum I have tried the LWIP mdns app. No luck, faced same problem during pairing process

mriksman, 2020-04-19:
Any of you using PWM?

mriksman, 2020-04-24:
Also, are you using ESP8266 RTOS SDK or esp-open-rtos?