maximkulkin/esp-homekit #117: General Discussion on Apple's move to open source part of Homekit

peros550, 2019-12-19:
Here is the link : https://github.com/apple/HomeKitADK

Do you think Apple's move would affect the future of this project?

maximkulkin, 2019-12-19:
So, here is my understanding of this: various chip manufacturers e.g. Espressif and Nordic offered HomeKit SDKs for their products and it looks like it was based on this thing. And I know that Espressif told to one of their clients that they can not make SDK for esp8266. Now it makes more sense to me: since it's unwise to modify Apple's ADK because it will make upgrades hard, most companies use it as is and just provide intermediate layer to adapt it to their platform (similar to FreeRTOS). And this makes it hard to optimize for particular hardware and thus no HomeKit on ESP8266 for you.

Unless someone volunteers to port ADK to ESP8266, this project for ESP8266 is fine.

On the other hand, since Apple open sourced their ADK, I expect hardware vendors like Espressif will be allowed to open source their ADKs too, so for ESP32 you might get Apple's ADK.

timoschilling, 2019-12-19:
Maybe it's possible to write an PAL for ESP8266
https://github.com/apple/HomeKitADK/blob/master/Documentation/PAL.md

closed 2020-01-27 by maximkulkin