maximkulkin/esp-homekit #119: Trying ESP_8266_RTOS_SDK parring trigger watchdog

wzc2019, 2019-12-24:
I'm trying ESP8266_RTOS_SDK to complie example\esp32\led.Then a compiled problem occured,I think wolfssl-4.1.0\wolfssl\wolfcrypt\port\Espressif\esp32-crypt.h is not  compatible but I don't know how it make and how to change. I have tried esp-wolfssl(ESP8266_RTOS_SDK's internal component),it short of a sha512.h file and failed to complie too.
Finally I tried wolfssl-3.13 and complied successfully.
But I met a problem when I flash the program and trying  to paring it with my phone.It took a long time and watchdog triggered . I have no idea how to solve it.
[image]

d4rkmen, 2019-12-24:
Works for me with wolfssl 4.1.0. Atlleast examples

maximkulkin, 2019-12-25:
If you get this error, try increasing watchdog timeout period (`Component config -> ESP8266-specific -> Task Watchdog timeout period (seconds)`)

closed 2020-01-27 by maximkulkin

mriksman, 2020-04-08:
> If you get this error, try increasing watchdog timeout period (`Component config -> ESP8266-specific -> Task Watchdog timeout period (seconds)`)

It's in (`Component config -> Common ESP-related -> Task Watchdog timeout period (seconds)`) as of ESP8266 RTOS SDK 3.3