maximkulkin/esp-homekit #120: Please add Status return value for char setter callback

d4rkmen, 2019-12-24:
In order to control operation status from setter. For example can send No answer status when sensor\anything else is unreachable. Or can return invalid value status for unsupported mode set
Thanks

maximkulkin, 2019-12-24:
How this is supposed to work?

d4rkmen, 2019-12-24:
process_characteristics_update:
HAPStatus status = HAPStatus_Success;
...
if (ch->setter_ex) {
                        status = ch->setter_ex(ch, h_value);
...
return status;

Usecase is very important: when user made an action, which cannot be completed on device side, setter can return error status and (Siri) will report - cant access your device. Othervise it will be success (but actually its not)
Lets say: your thermostat not supports cooling. User selected COOL and got success status.
Or an actor device is connected by some interface (serial/network/spi/i2c) which not responds - same problem.
For bridged acessories lost connection - same problem
Etc etc. Thanks for your attention.

closed 2026-01-03 by d4rkmen