maximkulkin/esp-homekit #121: What does "id" mean in HOMEKIT_ACCESSORY

Masterpc96, 2019-12-28:
I want to know what does this .id=1 mean 
  HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]){

For example if we have two accessories first must have .id=1 and second must have .id=2 ?

maximkulkin, 2019-12-28:
Kind of yes. Each accessory should have unique ID value. Also, if you push firmware update and you change your accessories, spec recommends that you preserve IDs of accessories, services and characteristics. Usually you do not assign IDs for services and characteristics, framework does it for you automatically, but if you change services/characteristics you probably want to set IDs explicitly.

Masterpc96, 2019-12-28:
Ok so it is for my knowledge only

closed 2019-12-29 by maximkulkin