maximkulkin/esp-homekit #124: SetupID and QR Code

mgoeller, 2020-02-07:
If I use the same SetupID and PIN Code for multiple devices, and power them up all at the same time (still unpaired) the pairing with a generated QR code fails. I guess this is supposed to be like this because the iOS device doesn't know with which device it should pair?

What is a good option to automate this procedure, e.g. having one firmware file but still being able to deploy it to multiple devices with the ability to pair them all with a QR code?

maximkulkin, 2020-02-13:
You need to use different QR codes for different devices. One options is to store setupId separate from firmware and flash it along with generic firmware. Then generate QR code for each accessory.

peros550, 2020-02-13:
What is Setup ID? Is it the PIN entered during pairing?

maximkulkin, 2020-02-13:
No, it's a 4 letter/number ID that is used in QR code (also, set .setupID in homekit_server_config_t)

mgoeller, 2020-02-17:
How would one store the setupID separate from firmware? Using sysparam and then set the setup ID during the flashing process at a predefined storage location?

maximkulkin, 2020-03-20:
I would opt for a tool that build a sysparam data on laptop and then flashes it, so when you read sysparams on chip the values are already there. Unfortunately, no such tool exists now and creating it is not an easy task to recommend. So, my suggestion is to pick an free address and write config there and then in firmware just read from a pre-defined location.

closed 2020-04-07 by maximkulkin