maximkulkin/esp-homekit #125: Problem after power outage

mgoeller, 2020-02-17:
Hi, I'm using your library for a lot of devices - thank you very much for that great piece of coding!

However now after a power outage for the whole house (stormy days in Switzerland ;)) I have three problematic devices (all of the ~20 devices are running exactly the same firmware), that show "no response" in Homekit, are not connected to the WiFi anymore (no ping) and when hooked up to serial show the following continues error message every .5 sec after booting up:

```
_task_hdl : 3fff0040, prio:14, stack:512
pm_task_hdl : 3ffef930, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3ff8, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Dec 27 2019 16:32:39
phy ver: 273, pp ver: 8.3

>>> wifi_config: Initializing WiFi config
>>> wifi_config: Connecting to EGHome //EGHome is my WiFi SSID
[W]sec 510 error
[W]sec 510 error
[W]sec 510 error
[W]sec 510 error
[W]sec 510 error
[W]sec 510 error
```

maximkulkin, 2020-02-17:
The error seem to be coming from [libmain](https://github.com/SuperHouse/esp-open-rtos/blob/master/open_esplibs/libmain/user_interface.c#L585-L600): looks like it is trying to save wifi data to flash (with verification) and verification fails and then it gets stuck in this loop. Seems like something is wrong with flash chip (or particularly one of last three sectors).

mgoeller, 2020-02-17:
Ok thank you, so I guess the flash was damaged with the power outage and there is no possibility then to replace (or write wifi data to another sector?).

AramVartanyan, 2020-02-17:
I am having the same issue on multiple devices. I do not think it is related to damaged hardware. It is always happening if the power supply is not stable or if the device is shut down in improper moment.
After reflash than erase_flash and second reflash, the same device is up and running again.

AramVartanyan, 2020-02-17:
This issue appeared after the update of esp-wifi-config (around September last year).

mgoeller, 2020-02-17:
It's good that I am not the only one. I can gladly provide a device and somehow help debugging, but have unfortunately no idea where to start.

peros550, 2020-02-17:
It has happened to me once a few weeks ago. As mentioned, just reflashing kept the stored pairing.

AramVartanyan, 2020-02-17:
> It has happened to me once a few weeks ago. As mentioned, just reflashing kept the stored pairing.

Yes. That is true, however I have noticed that you cannot erase_flash directly. (Shows 0.0 s and didn't actually erasing anything.) The flash memory falls in a strange state and the erase isn't working before you flash it at least once. So I prefer to do clean flash.

mgoeller, 2020-02-17:
> Yes. That is true, however I have noticed that you cannot erase_flash directly. (Shows 0.0 s and didn't actually erasing anything.) The flash memory falls in a strange state and the erase isn't working before you flash it at least once. So I prefer to do clean flash.

Interesting that you see the same behaviour there as well. Which device are you using? A Shelly as well? Did you notice the same problem with other ESP8266 based devices?

AramVartanyan, 2020-02-18:
I am using Shelly1 and Shelly 2.5 now. Both have the 510 error.
Sonoff basic devices I have are with older FW version and they work fine. Maybe I should try to reproduce the issue with them.

LuuTinhIT, 2020-03-15:
Same issue :( some devices always showing not responding. Waiting for fix this.

luisjimenez6245, 2020-04-20:
Same with me, i have to reset the device wifi config on every shutdown on the device or of my wifi network.

morphix9, 2020-05-11:
I'm using @RavenSystem's [ESP-Homekit-devices](https://github.com/RavenSystem/esp-homekit-devices) and have the exact same problem. Is there something new about that?

morphix9, 2020-05-18:
@maximkulkin can we hope on a fix for this issue? It's really painful to re-flash devices after almost every outage ;(

closed 2021-01-20 by mgoeller