maximkulkin/esp-homekit #126: Icon problem

CHNleing, 2020-02-28:
Why is the light bulb icon always displayed instead of the corresponding type icon when adding a device? Can it be customized?

d4rkmen, 2020-02-28:
@CHNleing 
`   HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]){
`
The icon depends of the category value

CHNleing, 2020-02-28:
thank you for your reply. Does it look like this.
HOMEKIT_ACCESSORY(.id = 10,.category = homekit_accessory_category_sensor,  ......

I have a question. What does setupId refer to and what is the content of it? Is it pure letters?

d4rkmen, 2020-02-28:
> .id = 10

id = 1 is the one will show up. The rest are bridged acessories

maximkulkin, 2020-02-28:
> What does setupId refer to and what is the content of it? Is it pure letters?

@CHNleing yes, setupID is digits and capital letters (esp-homekit will show you an error if you specify anything else). It is used to support pairing by QR code: you specify same setupID in accessory and when you generate QR code and you it will look for that accessory when you scan that QR code.

closed 2020-03-20 by maximkulkin