maximkulkin/esp-homekit #131: ESP RTOS sdk paring failed.

GoRo33, 2020-03-21:
Hi! 

First of all I want to express words of appreciation for the hard work put into this project. 

I have problem with connecting my iPhone to accessory. Apparently everything is fine, gets the connection but according to logs pairing is successful, while the phone shows information that the device cannot be added due to a connection error. 

Somebody could help to solve this issue? 

I'm using ESP RTOS SDK, branch release/v3.2 and iPhone 8 with latest iOS. 

```Bash
I (457) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
STA start
I (5247) event: sta ip: 10.52.0.18, mask: 255.255.255.0, gw: 10.52.0.1
WiFI ready
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 8A:5A:C8:14:AE:DB
>>> HomeKit: Configuring mDNS
>>> homekit_setup_mdns: Accessory Setup ID = 1QJ8
>>> homekit_run_server: Staring HTTP server
>>> HomeKit: Got new client connection: 1 from 10.52.0.19
>>> homekit_client_process: [Client 1] Got 131 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 49160
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 1] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 49068
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 1] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 48396
>>> homekit_server_on_pair_setup: [Client 1] Using user-specified password: 327-98-547
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Setting SRP verifier
>>> crypto_srp_get_public_key: Calculating public key
>>> send_tlv_response: [Client 1] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 3 value (384 bytes): \x03;\x91\xB5k!\xEC\xE6\x961w\x02\x03\x8D\xBB\x03\xA09\x9DQ\x8C\x97\x0A#\x92\x09<- o\xB5\xC5\x07s\xACR}\x01\x9Cm"]\xDB%D\x8E\xCD\x92^Q\xC6\xD4\xD2NQ\xCE,\x80\xD2S`\x05Y\xD1\xB5\xA6\xCA,\x95<{l\x09\x0E\xCC\xDC\x8F\xCD\xB2-\xCBN\xDE\x0F\xCD\xBEJt\xD75\x0F\x15\xB8>\x88u\xAF\xA9\xC3\xC0\x82\x1D\x13b\xF2\xC9613\x91\xE7\x88\xC6\xCFK-a`\x8F\x95\xD3#"\x84\x9Bg\x03\x08r\xC6\x1B\x91\xC7\x84\x1C\x9BY\xA1)5\x1D\\\x81\xD9_\xB2\x8B\xF5\x9FL\xC9!\xB5\xBF\xDE*&\xE9\x0AJw!\x12\x8F)\xE2;\x09\x9D\xD9\xBCvtL\xBE~W\xEF\x87\x17\xA1\xC7\xB68\x09\xD6\xA3\x04\xFD\x0E\x14\x11|O\xDE{\xC3\x9B\xC4\xEA\x11\x89\xC5\x07\x16\x89\xC9\x90\xA8\x11\xFA\xCC\xD5\xAF;9\xA2\xE9\x9B;'$;\x17\x9B\x18\xC7\x8DR\xCA\x87\x81\x0B\x06\xF3\xA438\xB6\xAE\xAC\xD4\xB8U\xD7~\xCC\xE2w\x0D\x08:"KT\xC6\xC8}\xE28Fx\xFF\xCDv \xC0\x9F\x9B\xFB/\x17\xBA@\x86\x1F!.\xF6\xB6\xEB\xF2\xF3\x0F\xB1}\x19:\x0CCO\x11\xB1\x12_\xCF\x82\xA0\x9Db\x8F^h*n\x9B5B'"4\xDED\x84w\x8F\x10\xA1uR\x97\xBE,\x81\x94{~\xAD\x1D\xB0]\xBC"\x83\xFC\xEE\xAB \xF7(\x8F\x10\xC9\x0BH=\\\x88Z\xF6\x1Bu\x04\xCD$\xBFi\xE1\xF0\x15D\x16\xF4\xC5\x1Dyi\x10b\x18{y\x831\xFF\xB2L\xCB\xBA\xCD>\xDD
>>> tlv_debug: Type 2 value (16 bytes): B\x16i\x01\xC6\x15\xFD"\x02\x87\x9BWR\xA6{\xDB
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 1] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 409\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x03\xFF\x03;\x91\xB5k!\xEC\xE6\x961w\x02\x03\x8D\xBB\x03\xA09\x9DQ\x8C\x97\x0A#\x92\x09<- o\xB5\xC5\x07s\xACR}\x01\x9Cm"]\xDB%D\x8E\xCD\x92^Q\xC6\xD4\xD2NQ\xCE,\x80\xD2S`\x05Y\xD1\xB5\xA6\xCA,\x95<{l\x09\x0E\xCC\xDC\x8F\xCD\xB2-\xCBN\xDE\x0F\xCD\xBEJt\xD75\x0F\x15\xB8>\x88u\xAF\xA9\xC3\xC0\x82\x1D\x13b\xF2\xC9613\x91\xE7\x88\xC6\xCFK-a`\x8F\x95\xD3#"\x84\x9Bg\x03\x08r\xC6\x1B\x91\xC7\x84\x1C\x9BY\xA1)5\x1D\\\x81\xD9_\xB2\x8B\xF5\x9FL\xC9!\xB5\xBF\xDE*&\xE9\x0AJw!\x12\x8F)\xE2;\x09\x9D\xD9\xBCvtL\xBE~W\xEF\x87\x17\xA1\xC7\xB68\x09\xD6\xA3\x04\xFD\x0E\x14\x11|O\xDE{\xC3\x9B\xC4\xEA\x11\x89\xC5\x07\x16\x89\xC9\x90\xA8\x11\xFA\xCC\xD5\xAF;9\xA2\xE9\x9B;'$;\x17\x9B\x18\xC7\x8DR\xCA\x87\x81\x0B\x06\xF3\xA438\xB6\xAE\xAC\xD4\xB8U\xD7~\xCC\xE2w\x0D\x08:"K\x03\x81T\xC6\xC8}\xE28Fx\xFF\xCDv \xC0\x9F\x9B\xFB/\x17\xBA@\x86\x1F!.\xF6\xB6\xEB\xF2\xF3\x0F\xB1}\x19:\x0CCO\x11\xB1\x12_\xCF\x82\xA0\x9Db\x8F^h*n\x9B5B'"4\xDED\x84w\x8F\x10\xA1uR\x97\xBE,\x81\x94{~\xAD\x1D\xB0]\xBC"\x83\xFC\xEE\xAB \xF7(\x8F\x10\xC9\x0BH=\\\x88Z\xF6\x1Bu\x04\xCD$\xBFi\xE1\xF0\x15D\x16\xF4\xC5\x1Dyi\x10b\x18{y\x831\xFF\xB2L\xCB\xBA\xCD>\xDD\x02\x10B\x16i\x01\xC6\x15\xFD"\x02\x87\x9BWR\xA6{\xDB\x06\x01\x02
>>> homekit_client_process: [Client 1] Finished processing
>>> homekit_client_process: [Client 1] Got 584 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 46208
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x03
>>> tlv_debug: Type 3 value (384 bytes): \xA5nsA\x13y\xEA\x15>\xEEXo*`\x0F\x09U\x1ERh\xEB\x09\x9DKA9\xA9\x82\x08/\xFA`\xB8\xA6!\xFCZ\x86\x05h\xCDt\xAE9q\xE2\x86\xF0\xB8\xEE\xE1\xA9}6\xF38\xA4vrST\xF0\xAF\xFDT\xB1"S\xDB\x9B\xE7A@\xD9\xD8>-\xF4\xD0\x07U\xD1\x8E&\xC4%\xCB\x83#d5m^ \xBCWLK+\xF3\x17U\x16\xA7m\xE3K\xA7=\xC9\x98\xD7\xCF\xCD\x89\xA7\x1AN\xFD\xA2\x12O~\xB5\x883$8ii5n\x16\xEAlEg\x95r\x8C\x86\xE3\x05\x8Dx\xA6\xC3\xC9\x08\\\x15\xF8k\xF2\x83\xA5\xD4\xF1\xBC=\xC4\x08\x0Ca\xC2\xFDd\x9D\xBE\xCE\x95b\xD8\xC3jkV\xF7SU\xFE\x97'0\x04\xAE\x1A\x12\xED\xF8w\xCD+\xF6\xEE#\xEC\x87\xBB_\xC7+^\xB4\x9A\x9F\xBA\xC3\xF7\x85\xE9\x84|\x19\xE8\xFFL\xC1!F\x01\xE9\xFA0\xC8\xB1"\xDF\xEB\xD4\x95\xE1\x11Mu\xF2\x0E\xD0^\xBE/\xFC8\xDA\x13f5\xC9wr\xEC\xA8+\xB2\xBD\x07\xFEL-\xA3\xFA\x10\x0Cq\xB3\xE3\x8B\xBD\xD7\x8A\x9F\x1C\xE9\x1C\xEBnp\xF1\x1B\xA4+\xF8\xD1\x9E?;N\xA9\\\x091Dq\x16\xCC\xE9\x1A\xF9G\x0AlI\x98\x83J\xF5\x07\xCA\xDB\xB9\x93\x8E\xD9\xF7\x04j\xF2H\x0B\x0D{\xC6\xA2\x09!\xF5!\x8C\x86\xF6\x9DaZ`j~\x00\xAA\x94^]&o\xAA\x13\xF7\x01\xA4\x8B\x88\xA2\xD5C\x9D\xF2\x15Qk-\x87\xB7E:hG\x83\x16\x008\xDA}\x81E {3I\xA5\xB3\\\xF6\x97\x00\x02
>>> tlv_debug: Type 4 value (64 bytes): \xD4\xCD\x1B.he\x0Eu\xFF\xAAH\x92WL\x8F\x8C9\x9Ahel\x00z\x92\x9BW/\xD6\xA7\xFC\xBC\xB0F\x8B\x01\xFB\x12\xB4Cz.\xCE\x8Dv\xD5\xDA\x82\x05\xDE\xBB\xBFl\xA9~\xC3!\xA6\x1F\xD1\xB2\xD4\xB6J
>>> HomeKit: [Client 1] Pair Setup Step 2/3
>>> homekit_server_on_pair_setup: Free heap: 45648
>>> homekit_server_on_pair_setup: [Client 1] Computing SRP shared secret
>>> homekit_server_on_pair_setup: Free heap: 45648
>>> homekit_server_on_pair_setup: [Client 1] Verifying peer's proof
>>> homekit_server_on_pair_setup: Free heap: 45968
>>> homekit_server_on_pair_setup: [Client 1] Generating own proof
>>> send_tlv_response: [Client 1] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x04
>>> tlv_debug: Type 4 value (64 bytes): \x09\x0DK$*\xC4\x87X}\x16\x1B\xC3\x9DS\x8DVf7=\x0DNg\xA2\x8F\x88\x1F\xB0\xA6\xAE\x93;n5"14M5\xBA;>kQ\xBD\x08J.\xF0p\x971b|J\x07O"\xB7G\xA2X\x00\xCEs
>>> client_send: [Client 1] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 69\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x04\x04@\x09\x0DK$*\xC4\x87X}\x16\x1B\xC3\x9DS\x8DVf7=\x0DNg\xA2\x8F\x88\x1F\xB0\xA6\xAE\x93;n5"14M5\xBA;>kQ\xBD\x08J.\xF0p\x971b|J\x07O"\xB7G\xA2X\x00\xCEs
>>> homekit_client_process: [Client 1] Finished processing
>>> homekit_client_process: [Client 1] Got 286 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 46828
[... 23 lines cut ...]
>>> tlv_debug: Type 5 value (135 bytes): U^D2\x83F\x06\x11\xB1\xE7\xE3X>\x80\xCAV\x88\xEF\xB4o\xE8\x97\xC0\xD4^xb\x06\xAD\xBF\xEF6|\xEA\x9D\xC1%Fw|iL&HZ3\x0B:\xD4\xCC8\xE5\x93A\xF3\x0B\xB4A\x1A\xA7"\xFF\xC4&1\xD2\xB0\x06\x85\x8D\xB8\x857z\x81\xEBD\x89\x00\xC43>-\x1B\xCB\xF4[x\xBBi\x8AE.\x0E?\x9C\x08EM\xB9\x1A3\x1B\x94\x9F\xF0\xD9Nj\xC8\xA3\xD9\xB6N\xEFH\xF2\xF0\x81q\xD41\x0F9\xC4Q\xC6\x9A\xF1\xF3L9\x1C
>>> client_send: [Client 1] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 140\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x06\x05\x87U^D2\x83F\x06\x11\xB1\xE7\xE3X>\x80\xCAV\x88\xEF\xB4o\xE8\x97\xC0\xD4^xb\x06\xAD\xBF\xEF6|\xEA\x9D\xC1%Fw|iL&HZ3\x0B:\xD4\xCC8\xE5\x93A\xF3\x0B\xB4A\x1A\xA7"\xFF\xC4&1\xD2\xB0\x06\x85\x8D\xB8\x857z\x81\xEBD\x89\x00\xC43>-\x1B\xCB\xF4[x\xBBi\x8AE.\x0E?\x9C\x08EM\xB9\x1A3\x1B\x94\x9F\xF0\xD9Nj\xC8\xA3\xD9\xB6N\xEFH\xF2\xF0\x81q\xD41\x0F9\xC4Q\xC6\x9A\xF [... cut ...]
>>> HomeKit: [Client 1] Refusing to pair: already paired
>>> send_tlv_response: [Client 1] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> tlv_debug: Type 7 value (1 bytes): \x06
>>> client_send: [Client 1] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 6\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x02\x07\x01\x06
>>> homekit_client_process: [Client 1] Finished processing
>>> HomeKit: [Client 1] Closing client connection
```

d4rkmen, 2020-03-21:
Hi, wolfssl random error I have too (every first boot after reflash) Not sure what the problem is. 

The pairing issue is well known (mDNS implementation + wrong place for paired flag set) and solving by PR #123

GoRo33, 2020-03-21:
Do you think changing branch to esp-RTOS-sdk 3.1 will help?

d4rkmen, 2020-03-21:
> Do you think changing branch to esp-RTOS-sdk 3.1 will help?

Definately, it will not help.

GoRo33, 2020-03-21:
> > Do you think changing branch to esp-RTOS-sdk 3.1 will help?
> 
> Definately, it will not help.

Ok, so only solution is to wait for PR to be merged ?

d4rkmen, 2020-03-21:
@GoRo33  Can you provide the log?

GoRo33, 2020-03-21:
Of cors. Happy to help: 
 I have made changes described in pull request commits add to components dir new mDns library and additionally turn off mDns library from menuconfig. 

```bash
STA start
I (23478) event: sta ip: 10.52.0.18, mask: 255.255.255.0, gw: 10.52.0.1
WiFI ready
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 82:16:56:B6:95:87
>>> HomeKit: Configuring mDNS
>>> homekit_run_server: Staring HTTP server
>>> HomeKit: Got new client connection: 1 from 10.52.0.16
>>> homekit_client_process: [Client 1] Got 131 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 54732
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 1] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 54632
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 1] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 53968
>>> homekit_server_on_pair_setup: [Client 1] Using user-specified password: 111-11-111
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Setting SRP verifier
>>> crypto_srp_get_public_key: Calculating public key
>>> send_tlv_response: [Client 1] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 3 value (384 bytes): \x8A\xB8\x80\xAF\x09\xC6\xB5]\xE1\xF7\x04$\xB2 kYA\xD0\xFE\x1B\xF6\xE0eQ=\xB7\xCFH\xB43\xD1\x12%\x9E\xADH/\x0C\x80@`?g\xB3\xD3\xA8w\xDB\xC8\x9D\xD9\xB5Q2\xEEbAW\x0F\x86\xFB\xF8\x15\x07\xAD\xFC\xE9\xA1n\xE0\xB4\x8E#\xDC$\x8EC\x81i\xE9\xE9\x9E^\x8C\xC5\x82s`{\x98s#Qun(\x9F7\xD5c\xF8\xB4\xBBH\xF1\xF7\x88k\x05%?Z\xA5\x8Bk\xEEh?5\xD1\xF0\xCCk.\xFF\x90fm\xE7p3\x97\x11\xF5w\xE7\xA0\xE1\x1D\xD2\xBEkPq\x82\x11\xEE\x9C\xCF\x9C\xD0H\xB8\x9F\x803k\xB3\xCFP\xBF\xDAX\xD6o\xCECa\xD4\xD0~\x8B\xF2\xD8\x1D\xE6\x91\x8FaI{;\xE0\x8F>\x80\x18z\x9B\xB59?\x05\xD4\x02t\x08#\x9D\xDF\xFB\x14n\xD8\xC7\xED\xA2\xF7\x05\x0B\x0E\xA2\xF1\x97\xDA\x84\xF7\x0C/]N\xDC#\x9FI\xEC\x86\x86\xBBh\x15\xE5\x8B\xB7\xE9\xA5\x9F\xD4\x81\xBC\xF4\xC4u\xB3\xB9\xF7T\xCD\xA7\x86\xA9\xC8\x83\xAF\xE3\xD2\x96gJ\x81\x1A\x95\xF7\xAF\x17i\xF2\xAA@>X'\x1DZ|\xC1\xA4\x02O0\x18}J;\xEE\xA3\xD3<\xD10 ~K\xD5O\x92\x8A\xD0\xC3n{`\x0B\xFA\xAC\xD5A\xC3\x8D\xBF\x81\x05\xFA+\xDDvJ\xA3K\xBEYW\x18\xFC\xB6\\\xECVO\x9E!\xC0\xECF\xD1\x01\xAD\xA4\xB2\xE5\x1A'\xBE\x17\x82\xB1:\x1B\x0C\xFA',H\x00\x89\x0B\xED\xF4\x94\xB1\xB7"\x06\xBEU\x86\xBF\x01\xE2v\xF9\xA5]j\x04\x86!}3\xE7\xB9\xD7
>>> tlv_debug: Type 2 value (16 bytes): \xCF\x97 \xC4\x0F\x1FH4\xAF\x91\xBA0P\xE0\xDD\x97
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 1] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 409\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x03\xFF\x8A\xB8\x80\xAF\x09\xC6\xB5]\xE1\xF7\x04$\xB2 kYA\xD0\xFE\x1B\xF6\xE0eQ=\xB7\xCFH\xB43\xD1\x12%\x9E\xADH/\x0C\x80@`?g\xB3\xD3\xA8w\xDB\xC8\x9D\xD9\xB5Q2\xEEbAW\x0F\x86\xFB\xF8\x15\x07\xAD\xFC\xE9\xA1n\xE0\xB4\x8E#\xDC$\x8EC\x81i\xE9\xE9\x9E^\x8C\xC5\x82s`{\x98s#Qun(\x9F7\xD5c\xF8\xB4\xBBH\xF1\xF7\x88k\x05%?Z\xA5\x8Bk\xEEh?5\xD1\xF0\xCCk.\xFF\x90fm\xE7p3\x97\x11\xF5w\xE7\xA0\xE1\x1D\xD2\xBEkPq\x82\x11\xEE\x9C\xCF\x9C\xD0H\xB8\x9F\x803k\xB3\xCFP\xBF\xDAX\xD6o\xCECa\xD4\xD0~\x8B\xF2\xD8\x1D\xE6\x91\x8FaI{;\xE0\x8F>\x80\x18z\x9B\xB59?\x05\xD4\x02t\x08#\x9D\xDF\xFB\x14n\xD8\xC7\xED\xA2\xF7\x05\x0B\x0E\xA2\xF1\x97\xDA\x84\xF7\x0C/]N\xDC#\x9FI\xEC\x86\x86\xBBh\x15\xE5\x8B\xB7\xE9\xA5\x9F\xD4\x81\xBC\xF4\xC4u\xB3\xB9\xF7T\xCD\xA7\x86\xA9\xC8\x83\xAF\x03\x81\xE3\xD2\x96gJ\x81\x1A\x95\xF7\xAF\x17i\xF2\xAA@>X'\x1DZ|\xC1\xA4\x02O0\x18}J;\xEE\xA3\xD3<\xD10 ~K\xD5O\x92\x8A\xD0\xC3n{`\x0B\xFA\xAC\xD5A\xC3\x8D\xBF\x81\x05\xFA+\xDDvJ\xA3K\xBEYW\x18\xFC\xB6\\\xECVO\x9E!\xC0\xECF\xD1\x01\xAD\xA4\xB2\xE5\x1A'\xBE\x17\x82\xB1:\x1B\x0C\xFA',H\x00\x89\x0B\xED\xF4\x94\xB1\xB7"\x06\xBEU\x86\xBF\x01\xE2v\xF9\xA5]j\x04\x86!}3\xE7\xB9\xD7\x02\x10\xCF\x97 \xC4\x0F\x1FH4\xAF\x91\xBA0P\xE0\xDD\x97\x06\x01\x02
>>> homekit_client_process: [Client 1] Finished processing
>>> homekit_client_process: [Client 1] Got 584 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 51788
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x03
>>> tlv_debug: Type 3 value (384 bytes): {\xD2\x93\xAF\x95\xA9\x05\xB1?\x02\x85^o\x0A\xCE\xF6\x17\xC0W\x83\x86m\x0D\x1A/\xEB\xBBO\x8E\xBBRHc`\xC5\xCB\x04-\xB7\xBA\x0B3\xAD\\\xC5\xDE'\x05\x09\x94\x9E/\x0D@^\xCC \\\xF5\xF1\x8D4\x87-h\xC4%\xA0\xA9\xAA\x0F\x18\x0C\x1A\x88\x06\xC7\x08`9\x1E\xB6\x94,\xB3\xC7\xC6gF\xB2$v\xA8o)\x06\xB0\x07\xEA?QR\x11\xF6E\xFE\x12Q\x9D\xFB\x1D\x11\x0Db\x82^\xA9U\x05\x1A\x0C\x8B\x8DM\xD9N\xF5\xA4\xBA)\x12\xBFl\xD5\xB3\xB1\xF1\x06\xDF-o4C.Z\xBF\x12\x1B\xAFI\xD3f\x06\xBC\x8C\xD8C\xFE\x8B]\xEBT\xB9\x11\x10\xCF\x0A\xA4\x04DxR<\xA7~_l\x0D\x10El}\x88E\xDA{\xFC\xB0_\x8B\xC2HS\xC8\xC4?\xD3\xD5\xE6;C\x8F6fA\xEA&\xF4\xB7z\x16z\x83\xA7!\xF8\x95\xC9\xEC\x93\xFB\xF8\x02\x8FG+\xE4\xED\x14\x1E\xDCh#\xDD'\xF5s\xCC'\xB1\x8F\x83\xDB\xBD\xAAxG\xAE2}\xB7\x95\xE8\xF2\x09\xC5\x13\x03*v,\xAF\xBA\xDE\x9Arsz\xC1\xEE4\xD0\xF5|\x03(yG\x1F\xB3\xF2k\xBA\x0E\xBA\x88<\x072\xA4>(S+\x90\xB1\xAC\xF1\xFF\x08\xE2\x05\x12\x86\x15\x10\xAALR\xDE\xD3\x0A\xE7\x03\x1A\x1A8(\xE6\xB4\xDF2&\x99\x88\ [... cut ...]
>>> HomeKit: [Client 1] Refusing to pair: already paired
>>> send_tlv_response: [Client 1] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> tlv_debug: Type 7 value (1 bytes): \x06
>>> client_send: [Client 1] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 6\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x02\x07\x01\x06
>>> homekit_client_process: [Client 1] Finished processing
>>> HomeKit: [Client 1] Closing client connection
```

d4rkmen, 2020-03-21:
This is same error. Paired flag is set, but to crypted message comes from the controller (iOS), This is the thing should be changed by PR. Check server.c file changes

GoRo33, 2020-03-21:
I have checked it and your changes ar present. I have also try to change in project git submodule from originally to yours fork (master branch) and recompile clean project. Nothing changes still error. Tomorrow I will try to make environment from open-rtos-sdk and try if this will work but this is compromise I prefer ESP sdk.

d4rkmen, 2020-03-22:
> open-rtos-sdk

I was move to ESP_RTOS because I have faced this issue. So, it will not help.
Checked the last log, yes, mDNS configuration was not performed right after "Successfully paired" and this is how I wanted it to be.
This mean, look closer to mDNS implementation. May be to add a few extra debug outputs. Every 1 second it should send announcement packet.
iOS side thinks accessory goes offline, thats why pairing context dropped and started all over.

GoRo33, 2020-03-22:
OK, I will try to debug this. But for now as I understand this library is not working with esp8266? Or is there any option to make it work? Like downgrade some components or make it on latest (master branch) esy-rtos-sdk ?

d4rkmen, 2020-03-22:
> library is not working with esp8266?

it works. Just pairing success 2-3 times of 10 for me (without code changes)

Can you try Eve app for pairing? Just for the test

It is strange the next pair setup try is performed after such a long delay (10 sec?)
All that time u see pairing in progress on iOS? Or error comes immediately? What message?

GoRo33, 2020-03-22:
I have done this from Eve and it's this same. 
Exacly as you describe. Paring is still in progres and after second connection attempt error shows: Paring Error, Unable to connect to device.

d4rkmen, 2020-03-22:
> End if you are connecting 10 times to Esp how you are reseting connected status on device?

I think I have patched the error message on already paired for debug. Can't remember now.

May be you try to remove the last commit from my PR? (the socet connection setup changes)

mriksman, 2020-04-08:
How did you get around this?

```
>>> HomeKit: Starting server
>>> HomeKit: Resetting HomeKit storage
>>> HomeKit: Formatting HomeKit storage at 0xd000
>>> HomeKit: Generated new accessory ID: 66:A6:DD:26:42:2E
Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.
Core 0 register dump:
PC      : 0x40228f27  PS      : 0x00000030  A0      : 0x40228f27  A1      : 0x3fff7c10
0x40228f27: hwrand_generate_block at c:\vscode\projects\esp8266_homekit\build/../components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/random.c:943
 (inlined by) wc_RNG_GenerateBlock at c:\vscode\projects\esp8266_homekit\build/../components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/random.c:883

0x40228f27: hwrand_generate_block at c:\vscode\projects\esp8266_homekit\build/../components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/random.c:943
 (inlined by) wc_RNG_GenerateBlock at c:\vscode\projects\esp8266_homekit\build/../components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/random.c:883
```

I'm getting a panic when I try and use;
```
#include <esp_system.h>

    static inline int hwrand_generate_block(uint8_t *buf, size_t len) {
        int i;
        for (i=0; i+4 < len; i+=4) {
            *((uint32_t*)buf) = esp_random();
```

When I try and use the software random generator;
```
    #include "esp_libc.h"

    static inline int hwrand_generate_block(uint8_t *buf, size_t len) {
        os_get_random(buf, len);
        return 0;
    }
```
I get a Watchdog Timeout panic.

How does one get this running on ESP8266 RTOS SDK??

mriksman, 2020-04-08:
I got mine working - without the panic when using `esp_random()` by using the following in `user_settings.h` instead of the current function.

```
    #include "esp_libc.h"

    static inline int hwrand_generate_block(uint8_t *buf, size_t len) {
        os_get_random(buf, len);
        return 0;
    }

    #define CUSTOM_RAND_GENERATE_BLOCK hwrand_generate_block
```

Which leads me to believe that initial panic/fail _IS_ what is causing the issue with pairing after rebooting.

mriksman, 2020-04-08:
Also, by removing
```
CURVE25519_SMALL
ED25519_SMALL
```
from CMakeLists.txt, you can almost halve the pairing time, at the expense of 70KB flash space.

I've added this to my CMakeLists.txt file for WolfSSL _and_ HomeKit;
```
if(CONFIG_HOMEKIT_SMALL)
  set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} \
    -DCURVE25519_SMALL \
    -DED25519_SMALL"
  )
endif()
```

mriksman, 2020-04-08:
How long does it take to pair for you? From the moment I type in all the 1's and it begins pairing, until I get the pairing complete message is about 32 seconds on this ESP8266 RTOS SDK. About 20 seconds if I disable the above.

maximkulkin, 2020-04-08:
I should say I see a lot of people doing builds with idf.py and having all kinds of problems. Just use Make, it will make your life so much easier.
People on Windows, sorry, I do not care about that OS, help yourself.

GoRo33, 2020-04-08:
@mriksman You are my king! I have manage to make it working after your change, so the problem was with random number generator in wolfssl. My paring time is some of 15 - 20 seconds because I have increase cpu clock to 160 Mhz. If you will be in Poland you will get a beer from me ;) 

@maximkulkin I'm building firmware only using make. And I don't like Window$ too ;)

closed 2020-04-08 by GoRo33

mriksman, 2020-04-08:
They may remove support for GNU Make. 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system-legacy.html
So it seems logical to keep up with the direction Espressif are heading. 

Don’t think using or not using Windows has much of an impact...?

mriksman, 2020-04-08:
@GoRo33 what’s odd is esp-homekit also uses esp_random() with no issues. So it must be the actual function itself in user_settings.h that’s causing the issue.

GoRo33, 2020-04-08:
@mriksman Maybe it is good idea to make a pull request with you solution? I'm sure that we are not only ones with this problem.