maximkulkin/esp-homekit #132: HTTP returns 302 for Fatal exception (28)

b18062a, 2020-04-04:
Hi,

I'm incredibly excited by what you're doing and all your tutorials! Thank you in advance! 

Please note, I've been working with ESP32 chips for some time, but always via the Arduino IDE, so this is my first real effort into a bigger world, and I apologize for my lack of knowledge and any frustration I might cause.

I'm using Windows to program my ESP01. Flashed the ESP01 with esptool.py in command prompt (admin) using "esptool.py -p /COM5 --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 otaboot.bin", and it writes and completes. The ESP01 boots and I get the config screen where I put in the details as you specify (I've done it over 10 times). As soon as I hit apply, I keep getting this over an over (see serial monitor details below). I've uninstalled esptool.py and reinstalled it. I've changed 1MB to 8m with no luck. I believe my ESP is the 1mb version (black board).

Any help would be greatly appreciated. (Also I read through this: https://github.com/maximkulkin/esp-homekit/issues/13, but couldn't get much out of it to help me).

"user-init-start
>>> wifi_config: Initializing WiFi config
>>> wifi_config: Found configuration, connecting to [name removed for GitHub]
user-init-done
mode : sta(cc:50:e3:08:c1:5a)
add if0
scandone
add 0
aid 4
cnt 

connected with [name removed for GitHub], channel 10
dhcp client start...
ip:192.168.1.14,mask:255.255.255.0,gw:192.168.1.1
>>> wifi_config: wifi_config_sta_connect_timeout_callback: Successfully connected
wifiready-done
--- ota_boot...0
OTABOOT VERSION: 1.0.0
--- ota_init
userbeta='0' otabeta='0'
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
active_sector: 0x0
--- ota_set_verify...OFF
--- DNS: done!
--- ota_get_version
--- ota_connect LocalPort=ce9f DNS IP:52.64.108.95 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
GET /HomeACcessoryKid/life-cycle-manager/releases/latest HTTP/1.1
Host: github.com

sent OK
HTTP returns 302 for Fatal exception (28): 
epc1=0x40240c84
epc2=0x00000000
epc3=0x4022ed94
excvaddr=0x00000000
depc=0x00000000
excsave1=0x402194b5
Registers:
a0 402194b5 a1 3fff8830 a2  00000000 a3  0000000d
a4  00000048 a5  0000ff00 a6  00000000 a7  0000004c
a8  3fff8c30 a9  00000400 a10 0000003b a11 00000000
a12 3fff1590 a13 00000000 SAR 0000001f

Stack: SP=0x3fff8830
0x3fff8830: 50545448 312e312f 32303320 756f4620
0x3fff8840: 0a0d646e 65746164 6153203a 30202c74
0x3fff8850: 70412034 30322072 31203032 38333a32
0x3fff8860: 2037353a 0d544d47 6e6f630a 746e6574
0x3fff8870: 7079742d 74203a65 2f747865 6c6d7468
0x3fff8880: 6863203b 65737261 74753d74 0d382d66
0x3fff8890: 7265730a 3a726576 74694720 2e627548
0x3fff88a0: 0d6d6f63 6174730a 3a737574 32303320

Free Heap: 27320
_heap_start 0x3fff2600 brk 0x3fffbbac supervisor sp 0x40000000 sp-brk 17492 bytes
arena (total_size) 38316 fordblks (free_size) 9828 uordblocks (used_size) 28488

 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
WA⸮l|⸮⸮⸮l⸮⸮⸮n⸮⸮n⸮⸮<⸮n⸮⸮#⸮n⸮⸮1⸮8n⸮⸮;pp_task_hdl : 3ffefac8, prio:14, stack:512
pm_task_hdl : 3ffef530, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4780, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Jan  5 2019 17:14:35
phy ver: 273, pp ver: 8.3"

HomeACcessoryKid, 2020-04-06:
this is caused by a change in GitHub webpage.
Please go to https://github.com/HomeACcessoryKid/life-cycle-manager, read up and start using version 1.9.11?

b18062a, 2020-04-06:
Thank you for the quick reply. I might have missed something, so I'm sorry in advance if I did, but no luck.

1. I had a read of that page. I saw the 'how to use it' section, but am clueless how to apply it as I'm using this: https://www.studiopieters.nl/esp8266-homekit-light-switch/.
2. In any case, I just ran git clone https://github.com/HomeACcessoryKid/life-cycle-manager.git, then I checked the changelog.md file and it shows information about 1.9.11 so I'd say I'm up-to-date.
3. I then erased and flashed the ESP01.
4. After booting I still get the wifi config page but after that, same issue, over and over.

Would I be right in guessing that the code needs to be updated on Achim's end (with his .bin files?)

Cheers,
Julius

ps. I used my Mac this time, where I managed to get the esp32-homekit-camera going last night. The issue is identical.

maximkulkin, 2020-04-07:
@b18062a I recommend you move this issue to [life-cycle-manager](https://github.com/HomeACcessoryKid/life-cycle-manager.git) repository as it belongs there.

closed 2020-04-07 by maximkulkin