maximkulkin/esp-homekit #133: Can't connect iphone

CHNleing, 2020-04-07:
!!! [ 554606] HomeKit: Failed to compact HomeKit storage: sector data read error
!!! [ 554613] HomeKit: Failed to write pairing info to HomeKit storage: max number of pairings
!!! [ 554621] HomeKit: [Client 1073683436] Failed to store pairing (code -2)

Using ESP8266-12E

maximkulkin, 2020-04-07:
From this piece of log there is so many things that are wrong, so my question is: can you try it on a different module?

pbendersky, 2020-04-07:
I got an error similar to this today when using CMake (I build using `idf.py build`). I got errors on writing flash when creating the device ID and key.

I reverted to `make` and it worked...

@CHNleing how did you build?
@maximkulkin are you aware of this issue or should I open another ticket?

CHNleing, 2020-04-07:
Probably because the default CPU frequency of arduino is 80mhz, it is normal after changing to 160mhz

maximkulkin, 2020-04-08:
I wonder how do you achieve this (to run ESP at 80MHz)

CHNleing, 2020-04-08:
I am not sure if Google Translate is accurate, I mean 80mhz will go wrong.  Normal with 160mhz

closed 2020-04-12 by CHNleing